embox/embox #1758: Extend `cpuinfo` command output

Deryugin, 2020-03-02:
This command is supposed to be somewhat like `lscpu` in GNU/Linux systems, which prints various information about CPU (number of cores, CPU family, cache size and so on).

Currently `cpuinfo` works on x86 platform (can be checked with `x86/qemu` template) and prints just vendor ID and time stamp counter value.

```
root@embox:/#cpuinfo                                                            
CPU Vendor ID:              AuthenticAMD
Current time stamp counter: 3728390192025691936
```

It would be nice to extend this info (for example, add CPU frequency and available extensions such as FPU/MMX/SSE). For x86 architecture this data could be obtained with `CPUID` instruction. Detailed description of the accepted arguments can be found in [Intel® 64 and IA-32 ArchitecturesSoftware Developer’s Manual](https://www.intel.com/content/dam/www/public/us/en/documents/manuals/64-ia-32-architectures-software-developer-vol-2a-manual.pdf), chapter "CPUID—CPU Identification", page 292.

checker5965, 2020-03-05:
@Deryugin Hi, I'm interested in taking this issue up. Could you please assign it to me?

Deryugin, 2020-03-05:
@checker5965 sure!

checker5965, 2020-03-05:
@Deryugin The CPUID instruction is returning incorrect (0) values in the registers for some cases. Do you have some suggestions?

Here is my code to get the frequency - 

`static void set_cpu_freq(struct cpu_info *info) {
	uint32_t reg[2]; /* registers: eax, ebx */
	
	asm volatile (	"movl $0x16, %%eax   \n\t"
					"cpuid				\n\t"
					"movl %%eax, %0		\n\t"
					"movl %%ebx, %1		\n\t"
					: "=g"(reg[0]), "=g"(reg[1])
					:
					: "%eax", "%ebx", "%ecx", "%edx");

	info->base_freq = reg[0];
	info->max_freq = reg[1];

	return;
}`

The `eax` and the `ebx` registers should return the base and max frequency respectively. However, I'm getting 0 as value for both.

checker5965, 2020-03-05:
@Deryugin I've solved this issue but it is untested - my processor does not support the leaf for frequency. As of now, I have written the code for base frequency, max frequency, FPU, MMX, and SSE. Should I add anything else?

Deryugin, 2020-03-06:
@checker5965 it's fine for the first issue :) I'll write some comments for your PR.

closed 2020-03-06 by Deryugin