embox/embox #1767: EPIT driver support

Deryugin, 2020-03-03:
EPIT (Enhanced Periodic Interrupt Timer) is a timer used in various i.MX boards.

This timer is emulated with QEMU 4.2.0 `sabrelite` machine, which works with `arm/sabrelite` template.

```
(qemu) info mtree
    ....
    00000000020d0000-00000000020d0fff (prio 0, i/o): imx.epit
    00000000020d4000-00000000020d4fff (prio 0, i/o): imx.epit
```

Technical details (register definition and so on) can be found in [i.MX 6Dual/6Quad Applications Processor Reference Manual](https://www.nxp.com/products/processors-and-microcontrollers/arm-processors/i.mx-applications-processors/i.mx-6-processors/i.mx-6quad-processors-high-performance-3d-graphics-hd-video-arm-cortex-a9-core:i.MX6Q?&tab=Documentation_Tab)

puranjaymohan, 2020-03-05:
I would like to work on this.
Please tell me how to get started with the drivers in embox.
Thanks

anton-bondarev, 2020-03-05:
You can use [pl031](https://github.com/embox/embox/tree/master/src/drivers/clock/pl031 ) as an example

puranjaymohan, 2020-03-06:
@anton-bondarev Hi,
How to test the driver, I have built the arm/sabrelite machine, but how should I include the driver and test it.
Thanks

anton-bondarev, 2020-03-09:
> @anton-bondarev Hi,
> How to test the driver, I have built the arm/sabrelite machine, but how should I include the driver and test it.
> Thanks

@puranjaymohan You can remove (or comment ) `include embox.driver.clock.cortexa9` from template and add you clock driver. If ticker cmd would work and timer test succeeds it will mean that it work correctly

closed 2020-03-09 by Deryugin