embox/embox #1768: Audio issues with auto_qemu script

Deryugin, 2020-03-03:
For some reasons qemu fails to use alsa for audio output when run with sudo.

However, we need sudo privileges to configure network, so we need to find out the way to have both network in audio with `auto_qemu` script.

## How to reproduce
1. `make confload-x86/qemu`
2. Add following lines to conf/mods.config 
```
	include embox.cmd.play
	include embox.driver.audio.intel_ac_pa(buffer_size=0x1f00)
	include embox.driver.audio.portaudio_lib
```
3. Add some WAV file to `conf/rootfs/` directory.
4. Run `./scripts/qemu/auto_qemu`

You'll see following errors:
```
XDG_RUNTIME_DIR (/run/user/1000) is not owned by us (uid 0), but by uid 1000! (This could e.g. happen if you try to connect to a non-root PulseAudio as a root user, over the native protocol. Don't do that.)
ALSA lib pcm_dsnoop.c:641:(snd_pcm_dsnoop_open) unable to open slave
alsa: Could not initialize ADC
alsa: Failed to open `default':
alsa: Reason: No such file or directory
```

And if you try to `play your_file.wav`, more errors will follow.

Looks like this is caused by running qemu with sudo.
If you run qemu like this
```
qemu-system-i386 -kernel ./build/base/bin/embox -m 1024 -soundhw ac97 -nographic
```
Then there are not errors and you can play audio file.

But for
```
sudo qemu-system-i386 -kernel ./build/base/bin/embox -m 1024 -soundhw ac97 -nographic
```

It's same error messages again.

```
> qemu-system-i386 --version
QEMU emulator version 4.2.0
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers
```

shivamanipatil, 2020-03-03:
Hey, I did not get any errors while performing the first 4 steps and then playing a .wav file. I have attached some screenshots. I am using **Debian 9** and qemu version - **1:2.8+dfsg-6+deb9u9**.
[image]
[image]
[image]

Deryugin, 2020-03-04:
@shivamanipatil can you try it with newer qemu, like 4.0+? I guess that will help to find out if it's caused by qemu or Linux sound system.

sk2sat, 2020-03-09:
This issue also happed in my environment(QEMU 4.2.0 on Arch Linux).
I think this problem is due to running QEMU as root.
In addition, there is a warning on the Arch wiki, `QEMU should never be run as root`.
https://wiki.archlinux.org/index.php/QEMU#Running_virtualized_system
Therefore, adding `-runas` option solved this issue.
PR is here: #1809

PR #1809 (2020-03-11): scripts: Fix QEMU run as root
fix #1768
closed 2020-03-11 by Deryugin