embox/embox #1769: Implementing sp804 driver

Deryugin, 2020-03-03:
sp804 is a timer module which is used in various ARM boards.
Technical reference manual can be found [here](http://infocenter.arm.com/help/topic/com.arm.doc.ddi0271d/DDI0271.pdf).

It's emulated for qemu vexpress machine, which is already supported (`arm/vexpress-a9` template), so it's required to implement a driver for this timer.

```
(qemu) info mtree
    ...
    0000000010011000-0000000010011fff (prio 0, i/o): sp804
    0000000010012000-0000000010012fff (prio 0, i/o): sp804
```

layman-n-ish, 2020-03-03:
Hello! I see that LKDDb (Linux Kernel Driver DataBase) has [support for SP804](https://cateee.net/lkddb/web-lkddb/ARM_TIMER_SP804.html). May I take that as a reference and give an attempt at the implementation? Is there any other starting point you'd suggest?

anton-bondarev, 2020-03-04:
Yes, I've assigned the issue with you

alexkalmuk, 2020-03-05:
> May I take that as a reference and give an attempt at the implementation? Is there any other starting point you'd suggest?

@layman-n-ish You can also take a look at our src/drivers/clock, most of them are quite simple.

layman-n-ish, 2020-03-07:
Hi! SP804 is a **dual-timer** module; I see no other way than initialising two distinct `clock_source` devices for each timer because each timer would have its own `TARGET_HZ`, `IRQ` line, `read()` function etc. Am I right?

alexkalmuk, 2020-03-07:
> Hi! SP804 is a **dual-timer** module; I see no other way than initialising two distinct `clock_source` devices for each timer because each timer would have its own `TARGET_HZ`, `IRQ` line, `read()` function etc. Am I right?

Hm, I am not sure, but couldn't we use only one of these internal timers?

puranjaymohan, 2020-03-09:
> Hi! SP804 is a **dual-timer** module; I see no other way than initialising two distinct `clock_source` devices for each timer because each timer would have its own `TARGET_HZ`, `IRQ` line, `read()` function etc. Am I right?

Yes, EPIT is also similar to this timer and has two distinct timers in it with different registers. I think the current situation requires only a single timer as it is being used for frequency generation. I think you should only use a single timer.

Thanks

layman-n-ish, 2020-03-10:
Yes, that makes sense. Thanks, @puranjaymohan.

EugenySvirin, 2021-02-19:
I would like to try to finish this issue.

commit e3527b9e (2021-02-23): driver: clock: Add driver support for SP804

This adds driver support for ARM Dual-Timer Module (SP804)
which is emulated in qemu-vexpress-a9.

This closes #1769.
closed 2021-02-23 by anton-bondarev