embox/embox #1771: Implement some pthread_rwkock_xxx routines

anton-bondarev, 2020-03-03:
There are not implemented routines return -ENOSYS
```
int pthread_rwlock_tryrdlock(pthread_rwlock_t *rwlock) {
	return -ENOSYS;
}
int pthread_rwlock_trywrlock(pthread_rwlock_t *rwlock) {
	return -ENOSYS;
}
int pthread_rwlockattr_destroy(pthread_rwlockattr_t *attr) {
	return -ENOSYS;
}

int pthread_rwlockattr_getpshared(const pthread_rwlockattr_t *rwlock, int *pshared) {
	return -ENOSYS;
}

int pthread_rwlockattr_init(pthread_rwlockattr_t *rwlock) {
	return -ENOSYS;
}

int pthread_rwlockattr_setpshared(pthread_rwlockattr_t *rwlock, int pshared) {
	return -ENOSYS;
}

int pthread_rwlock_init(pthread_rwlock_t *rwlock, const pthread_rwlockattr_t *attr) {
	if (attr != NULL) {
		return -ENOSYS;
	}

	rwlock_init(rwlock);
	return 0;
}
```

subhojit-ghorai, 2020-03-08:
Hi! Can I work on this issue?

MrSquanchee, 2020-03-09:
@Deryugin  Hii I have done some research for pthread APIs in linux. I would like to work on this issue :) 
Thanks,
Suraj.

anton-bondarev, 2020-03-09:
@MrSquanchee Hi, Suraj
I've added another issue for pthread routines (#1805)
Can you please work on it?

anton-bondarev, 2020-03-09:
@subhojit-ghorai
I've assigned it to you

MrSquanchee, 2020-03-09:
> @MrSquanchee Hi, Suraj
> I've added another issue for pthread routines (#1805)
> Can you please work on it?

Sure. I have commented on the mentioned issue.
Thanks.