embox/embox #1774: questions regarding Mybuild syntax

ThePiGrepper, 2020-03-03:
a few questions regarding Mybuild: what do these do?
- static module <name> vs module <name> ?
- @DefaultImpl(module_name) (does this force module_name to be added to the build? what would happen if module_name didnt extend an abstract module, if it were a regular module?
- @NoRuntime depends
- @App
- @InitFS
- is the order in which modules are declared on mods.config of any relevance? if so, something like a @BuildDepends() can fix the potential issue of having modules in the wrong order?
- is there a difference in the build process between having, for instance, Dash being compiled with all its module dependencies defined inside its Mybuild file and having to externally write a couple of them in the mods.config file(as is the case now)?

*Feel free to close this issue after answering the questions. Another idea would be to keep it open and turn it into a draft/source of the documentation.

ThePiGrepper, 2020-03-04:
I'll start replying and filling the information as I'm getting to know the system. Maybe someone else will find this useful.
- Static modules are static libraries, they are just linked to the embox image. modules are object files which can be also initialized at run time.
- @Runlevel(num): It gives a recommendation to the build system for it to build the module in a certain order.
- @BuildDepends: It forces the particular module to be built after the modules specified inside BuildDepends annotation.
- @NoRuntime: prepends 'depends', the same as depends, without claiming an initialization order.
- @InitFS(): exports specified file (source) into embox's initfs as part of rootfs.cpio archive.
optional parameter: target_dir="directory"
- @App: it makes a module look like a loadable application. It signals that such a module will need to initialize correctly all its sections(ie: .bss will be zeroed,etc).
- @Autocmd: Makes it possible to statically link non-library modules with their own main() to the embox build. It's used for cmd line utilities.

alexkalmuk, 2020-03-04:
> @runlevel(num): It gives a recommendation to the build system for it to build the module in a certain order.

Not `build`, but `initialize` at run time.

ThePiGrepper, 2020-03-04:
> > @runlevel(num): It gives a recommendation to the build system for it to build the module in a certain order.
> 
> Not `build`, but `initialize` at run time.

oops. sorry. fixed

ThePiGrepper, 2020-03-16:
One more question. what does @IncludeExport(path="util") do exactly? I'm having problems understanding how the include structure works with the build system.

alexkalmuk, 2020-03-17:
> One question. what does @IncludeExport(path="util") do exactly? I'm having problems understanding how the include structure works with the build system. Also, the other parameter available, target_name=, how does it work?
> 
> EDIT: I figured it out. It's a modifier of the next 'source' command, instead of moving it to its relative directory in build/, it moves it to /build/include/${path}. 'target_name' changes the name of the exported file to ${target_name}.

```
module foo {
@IncludeExport(path=<path>, target_name="name.h")
source "foo.h"

source "foo.c"
}
```

The one can use it in other modules which depends on `foo`:

```
#include <path/name.h>
```
Note: `<target_name>` is optional, if it is not specified the actual header name (`foo.h`) will be used.

anton-bondarev, 2020-03-26:
@287mdsahil 
In the file 
third-party/qt/plugins/platforms/Emboxfb.my
I am having difficulty understanding the what the following line does:
@Build(stage=2,script="TARGET=$(EXTERNAL_BUILD_DIR)/third_party/qt/core/qt-everywhere-opensource-src-4.8.5/src/plugins/platforms/emboxfb; $(EXTERNAL_MAKE_PRO)")

Can anyone please explain its purpose

anton-bondarev, 2020-03-26:
@287mdsahil
`@Build` is an annotation which add build rules to a module.  Available 'stage' and 'scripts' option.
 'stage' says that the module is build on which step of project building the module was build. stage=0 is for internal Embox module, stage=1 usually for external libraries, and stage=2 is for external libraries or applications which has external dependencies.
'script' says what external script must be called before build module, for example (./configure; make; make install) use often in the project. In this example, the original Qt project file (.pro) will be used for building.

287mdsahil, 2020-03-27:
What does `@Generated` and `@AddPrefix` do?

anton-bondarev, 2020-03-28:
@287mdsahil 
'@AddPrefix' changes default search sources path.
For example
```
	@AddPrefix("^BUILD/extbld/third_party/games/quake3/release-linux-x86_64/")
	source "ioquake3.x86_64.a"
```
`"ioquake3.x86_64.a"` will be looked for `""^BUILD/extbld/third_party/games/quake3/release-linux-x86_64/"` directory
It can be required when you build external modules.

@Generated
Allow to execute any scripts to generate sources
For example
```
	@Generated(script="$(MAKE) -f $(ROOT_DIR)/mk/script/qt-plugin.mk MK_PLUGIN_NAME=emboxfb")
	source "emboxfb.cpp"
```
For  example
There is no "emboxfb.cpp" at first
Execute "$(MAKE) -f $(ROOT_DIR)/mk/script/qt-plugin.mk MK_PLUGIN_NAME=emboxfb"
which generate it and then build system will use generated file