embox/embox #1778: Wrong time in initfs

anton-bondarev, 2020-03-04:
stat returns the default value(0) always for initfs
```root@embox:/#stat ./about.html                                                  
  File: ./about.html
  Size:      565    Blocks:        0    IO Block:        0  regular file
   Dev:        0     Inode:        0       Links:        0
Access:    33188       Uid:        0         Gid:        0
Access:        0  
Modify: 1583322210  
Change:        0
```
times can be saved during initfs creating

ThePiGrepper, 2020-03-04:
I assume Modify was set during the initfs creation. What exactly would be the issue? Not updating then, or not updating after, let's say, a touch(access and modification) or cat(access)?

anton-bondarev, 2020-03-05:
> I assume Modify was set during the initfs creation. What exactly would be the issue? Not updating then, or not updating after, let's say, a touch(access and modification) or cat(access)?
> 
> edit: apparently cat doesnt update last access time, my bad.
> edit #2: also, now that we're talking about stat, and last access/data/status change/data modification fields, I'm curious. Are we staying by default with a POSIX-compliant behaviour (strictatime), or a more 'lazy' approach(relatime)? that will be important at the time of checking this issue probably.

In my mind. initfs should be kept read only and so couldn't be modify by cat or touch. So I mean that access and change times (now zero) can be set when initfs image is creating. And in runtime we can get it time

anton-bondarev, 2020-03-05:
> Are we staying by default with a POSIX-compliant behaviour (strictatime), or a more 'lazy' approach(relatime)? that will be important at the time of checking this issue probably.

I prefer POSIX-compliant behaviour (strictatime), but it applies not read-only file systems (not initfs), may be FAT on usb or hd. but it the different task, I think

ThePiGrepper, 2020-03-09:
I did some tests and apparently something similar is happening with vfat and touch/stat as well. I'll test again and report it here. Just to confirm then, what do you need initfs to do, and what is it doing now? I'd like to fix this issue if possible.