embox/embox #1781: Add USB test

alexkalmuk, 2020-03-05:
There are OHCI and EHCI support for USB, which currently verified by hands as described here - https://github.com/embox/embox/wiki/Using-USB-on-qemu

How we can automate this to add the corresponding test?

* Running `block_dev_test` on a USB flash drive would be enough (maybe through scripts/expect). The advantage of using `block_dev_test` is that we can easily run it on any device, not only qemu.
* Another way is to prepare a flash drive with a predefined data, then add a unit test (src/tests) that reads this block device and calculates md5sum and checks whether it is correct.
* ... more ideas?

btw, it's related not only to USB, but any other block devices, but I would propose to start with USB just to check these ideas :)

p.s.
After we will automate USB read/write testing, we can also automate USB flash drive insertions/deletion using QEMU's monitor `dev_attach/dev_detach`, but it is another issue..

rushilkvs, 2020-03-07:
Hi, This looks interesting. I would like to work this.

alexkalmuk, 2020-03-07:
Yes, you can get a try, but this particular issue is more related to shell scripts like scripts/continuous/, and expect - scripts/expect I think. But if you are interested in this then well.

alexkalmuk, 2020-03-08:
I was thinking a bit about that and probably one of the simplest solutions is just to run qemu with usb flash drive and add block_dev_test into the start script.