embox/embox #1785: Interactive pingb test false positive result

AntonKozlov, 2020-03-05:
https://github.com/AntonKozlov/embox/runs/488694733?check_suite_focus=true#step:6:208
When ping fails it reports something like
```
> ping 8.8.8.8 
PING 8.8.8.8 (8.8.8.8) 64 bytes of data
--- 8.8.8.8 ping statistics ---
4 packets transmitted, 0 received, +0 errors, 100% packet loss, time 4009ms
ping: Command returned with code 4: Unknown error code. PLEASE REPORT ISSUE 519!
```
Now interactive ping test assumes success if `0 errors` string is found in ping output https://github.com/embox/embox/blob/master/scripts/expect/ping.exp#L22

The test passes when ping fails.

checker5965, 2020-03-06:
@AntonKozlov What if we change the test case to also look for "0 received"? If ping fails we won't be receiving any packets.

AntonKozlov, 2020-03-06:
I haven't thought carefully enough at the issue. It should be a robust regex, I may suggest taking a full line `4 packets transmitted, 0 received, +0 errors, 100% packet loss`. But I haven't thought much about it

Another option is to consider return value. It's less error-prone and it allows noninteractive tests

alexkalmuk, 2020-03-06:
>  I may suggest taking a full line 4 packets transmitted, 0 received, +0 errors, 100% packet loss

No, even one non-replied packet should be considered like a failure (`75% packet loss` should be a failure too). I think `4 packets transmitted, 4 received` would be better.

Anyway, interactive tests aim to test command's _output_, not only command result, so I am agree ping is not the case.

As far as I can see we already have ping tests in `templates/x86/test/net/start_script.inc`:
```
"ping 10.0.2.10",
"ping 8.8.8.8",
"ping google.com",
```
so we can remove ping.exp for sure. I'd propose only preserve the test for ping flooding and add it just to scripts/continues/run.sh.

@AntonKozlov what do you think?

AntonKozlov, 2020-03-06:
> > I may suggest taking a full line 4 packets transmitted, 0 received, +0 errors, 100% packet loss
> 
> No, even one non-replied packet should be considered like a failure (`75% packet loss` should be a failure too). I think `4 packets transmitted, 4 received` would be better.

Damn, I copy-pasted wrong line. 100% agree, need to match "all replies received".
Although, it doesn't tolerate to network errors, and even "one reply recieved" may be success in unstable networks -- but I don't think this is our case with `google.com`

> As far as I can see we already have ping tests in `templates/x86/test/net/start_script.inc`:
> 
> ```
> "ping 10.0.2.10",
> "ping 8.8.8.8",
> "ping google.com",
> ```
They are simple and easy to check by eye, they supposed to provide basic sense that network is usable.
But if these fail (for some reason), embox fails to boot and many other tests are blocked.They are not flexible, I had to choose ip/hostname to ping on test start and cannot do this without kind of `sed` that can be avoided with external test suite.

> so we can remove ping.exp for sure. I'd propose only preserve the test for ping flooding and add it just to scripts/continuous/net/run.sh.

I'm not sure. I suggest to 
1. fix the `ping.exp` at least (that should be not hard)
2. remove duplicates `ping.exp`, `ntpdate.exp` and infra

And only later consider removing `ping.exp` or removing other pings besides `10.0.2.10` (the only one that is always available) from start_script

alexkalmuk, 2020-03-06:
> Although, it doesn't tolerate to network errors, and even "one reply recieved" may be success in unstable networks -- but I don't think this is our case with google.com

If it would happen because google.com do not response, we can either re-launch test or fix google.com to another url, or just get rid of this idea at all. But if we would allow embox to receive only one of four ICMP replies and treat this like success than accidentally we will mask possible network bugs in embox. I mean what if it is a bug inside embox, not the network instability?

> I had to choose ip/hostname to ping on test start and cannot do this without kind of sed that can be avoided with external test suite.

Ah, then ok, we can fix ping.exp then, but ideally ping shouldn't be tested interactively of course...

> remove duplicates ping.exp, ntpdate.exp and infra

What do you mean? ping.exp and ntpdate.exp duplicates?

AntonKozlov, 2020-03-06:
> If it would happen because google.com do not respond, we can either re-launch test or fix google.com to another url, or just get rid of this idea at all.

I don't want to write this kind of tests in start_script :) 

> But if we would allow embox to receive only one of four ICMP replies and treat this like success than accidentally we will mask possible network bugs in embox. I mean what if it is a bug inside embox, not the network instability?

I perceived this as functional test: we can resolve the name, handle gateway, send request, and get reply.

You are talking about different type of testing (reliability?). It should be performed against 99.(9)% reliable peer -- the host itself. 
I don't feel enough confidence after receiving 1 vs 4 replies from google.com

> > remove duplicates ping.exp, ntpdate.exp and infra
> 
> What do you mean? ping.exp and ntpdate.exp duplicates?

There are two problems described in the issue, I changed the description in a half-hour after issue was created. scripts/expect/ping.exp vs scripts/autotest/testsuite/net/ping/test.exp

alexkalmuk, 2020-03-06:
> I perceived this as functional test: we can resolve the name, handle gateway, send request, and get reply.

> You are talking about different type of testing (reliability?). It should be performed against 99.(9)% reliable peer -- the host itself.
I don't feel enough confidence after receiving 1 vs 4 replies from google.com

I do not say 1 vs 4 replies, I just want to make sure it will be _all_ replies, not 1 or 3, or 42. It could be only 1 ICMP packet to send somewhere to check this, I agree, but it should be stable (1 send, 1 received on every launch).

> There are two problems described in the issue, I changed the description in a half-hour after issue was created. scripts/expect/ping.exp vs scripts/autotest/testsuite/net/ping/test.exp

It's another issue I guess, `scripts/autotest` is just an experimental version which is intended to be more convenient to use and also to replace `scripts/expect` in the future.

alexkalmuk, 2020-03-06:
I am already not sure what we are discussing :)

1. How we can check ping works (from embox)?
As I understand you do not want to use google.com. So what do you propose? To create a local network with QEMUs or something else?
2. How to write a test for that
As far as I can see you proposed to fix current ping.exp or am I missed something?

alexkalmuk, 2020-03-06:
@AntonKozlov I mean if you stuck a bit on Github Actions with `ping google.com` you can just disable it, and we can use it only on Travis.

So one of the possible solutions is to fix ping.exp so it would correctly check whether one single ICMP packet was successfully transmitted and reply received from 8.8.8.8. Also, we can use an environment variable for this IP address, as you suggested before. And this test will be available only on Travis.

What about this? I'm just trying to understand how we can help now with the issue.

AntonKozlov, 2020-03-06:
> I do not say 1 vs 4 replies, I just want to make sure it will be _all_ replies, not 1 or 3, or 42. It could be only 1 ICMP packet to send somewhere to check name resolution, gateway etc., I agree, but it should be stable (1 send, 1 received on every launch).

This is impossible in networks in general. Intermediate nodes are dying. Servers overloaded with other requests.
But getting <100% replies from google.com for requests sent with 1 sec interval is such rare, that I'm fine to consider it functional test.

> I am already not sure what we are discussing :)
> 1. How we can check ping works (from embox)?
>    As I understand you do not want to use google.com. So what do you propose? To create a local network with QEMUs and other stuff or something else?

This is definitely out of scope of 1) ping expect test doesn't test anything 2) there are two expect tests for ping, ntpdate, probably more

> 2. How to write a test for that
>    As far as I can see you proposed to fix current ping.exp or am I missed something?

I _suggested_ the fix :)

> @AntonKozlov I mean if you stuck a bit on Github Actions with `ping google.com` you can just disable it, and we can use it only on Travis.

I think I have this resolved. But I'm wonder why are we discussing GHA in context of obviously wrong ping test. Although I originally found the issue while transferring to GHA. This ping expect test mislead me and made think ping works while it doesn't.
But please consider GHA is just context why have I looked at expect tests at all.

> So one of the possible solutions is to fix ping.exp so it would correctly check whether one single ICMP packet was successfully transmitted and reply received from 8.8.8.8. Also, we can use an [...]

Wait. I thought your statement was success is 4 of 4 replies from some google server with interval 1sec. I'm fine with this. 
I don't need the variable right now. Without this issue fixed expect tests may ping even my home router and I still would be fine (my router doesn't respond for pings :))

alexkalmuk, 2020-03-06:
> This is impossible in networks in general. Intermediate nodes are dying. Servers overloaded with other requests.
But getting <100% replies from google.com for requests sent with 1 sec interval is such rare, that I'm fine to consider it functional test.

I understand that, and that's why you cannot provide 100% stable tests at all!

alexkalmuk, 2020-03-06:
And no matter network or something else

alexkalmuk, 2020-03-06:
> Wait. I thought your statement was success is 4 of 4 replies from some google server with interval 1sec. I'm fine with this.
I don't need the variable right now. Without this issue fixed expect tests may ping even my home router and I still would be fine (my router doesn't respond for pings :))

Ok, I will try to fix the test, and that's all fotr the issue.

alexkalmuk, 2020-03-06:
@AntonKozlov Thank you a lot for the report! I would be also glad if you review https://github.com/embox/embox/pull/1793

closed 2020-03-07 by alexkalmuk