embox/embox #1791: risc-v: Slow time

alexkalmuk, 2020-03-06:
template: riscv/qemu-sifive-u
qemu: 4.2.0

```
make confload-riscv/qemu-sifive-u
```

Then add `include embox.cmd.testing.ticker` to conf/mods.config and do `make`.

```
$ qemu-system-riscv32 -nographic -machine sifive_u -kernel build/base/bin/embox

***

root@embox:/#ticker                                                             
1 sec
2 sec

```

alexkalmuk, 2020-03-06:
@zhuanyanchengkong Please let's know if you will start solving this issue :)

zhuanyanchengkong, 2020-03-07:
> @zhuanyanchengkong Please let's know if you will start solving this issue :)

OK,Thanks, I started to resolve this issue.

alexkalmuk, 2020-03-08:
@zhuanyanchengkong Update your fork please from the embox upstream, there were fixes related to risc-v qemu 4.2.0.

zhuanyanchengkong, 2020-03-08:
> @zhuanyanchengkong Update your fork please from the embox upstream, there were fixes related to risc-v qemu 4.2.0 (#1653)

Ok

zhuanyanchengkong, 2020-03-08:
> @zhuanyanchengkong Update your fork please from the embox upstream, there were fixes related to risc-v qemu 4.2.0 (#1653)

I have entered embox and reproduced the problem.

alexkalmuk, 2020-03-08:
> > @zhuanyanchengkong Update your fork please from the embox upstream, there were fixes related to risc-v qemu 4.2.0 (#1653)
> 
> I have entered embox and reproduced the problem.

Great! So now you can examine why this slowdown happens.

zhuanyanchengkong, 2020-03-09:
> Great! So now you can examine why this slowdown (during `ticker`) happens.

I think the most direct way is `#define HZ = 10000` in drivers/clock/riscv_clk.c.

alexkalmuk, 2020-03-09:
> > Great! So now you can examine why this slowdown (during `ticker`) happens.
> 
> I think the most direct way is `#define HZ = 10000` in drivers/clock/riscv_clk.c.

I do not think you should change HZ because `HZ=1000` just says the clock will be generated 1000 times per one second. So you should found in drivers/clock/riscv_clk. how to change an actual frequency. Look,

```
#define COUNT_OFFSET (SYS_CLOCK / HZ)
```

So you can see that this COUNT_OFFSET actually depends also on `SYS_CLOCK`

SYS_CLOCK is defined in src/include/hal/system.h, and you can see it is an option...

zhuanyanchengkong, 2020-03-09:
> SYS_CLOCK is defined in src/include/hal/system.h, and you can see it is an option...

Yes.We can modify the `core_freq` in  `templates/riscv/qemu-sifive-u`

> include embox.arch.system(core_freq=10000000)

Then the clock works correctly．

anton-bondarev, 2020-03-09:
> > SYS_CLOCK is defined in src/include/hal/system.h, and you can see it is an option...
> 
> Yes.We can modify the `core_freq` in `templates/riscv/qemu-sifive-u`
> 
> > include embox.arch.system(core_freq=10000000)
> 
> Then the clock works correctly．

Can you create PR with modified core_freq?

zhuanyanchengkong, 2020-03-09:
> Can you create PR with modified core_freq?

Yes！My pleasure.

anton-bondarev, 2020-03-26:
Closed #1861

closed 2020-05-16 by anton-bondarev