embox/embox #1799: x86 ACPI is broken

alexkalmuk, 2020-03-07:
Template: `x86/qemu`

Add module `include embox.arch.x86.kernel.acpi_shutdown` to the mods.config, then `make`.

```
gcc  -pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -std=gnu99 -O0 -gdwarf-2 -nostdinc -m32 -march=i386 -fno-stack-protector -Wno-array-bounds -Wno-misleading-indentation -Wno-nonnull-compare -U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros build/base/gen/config.lds.h -Ibuild/base/include -I./src/include -I./src/arch/x86/include -Ibuild/base/gen/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/include -Ibuild/base/gen/src/arch/x86/include  -I./src/compat/linux/include -I./src/compat/posix/include -I./src/compat/libc/include -I/home/alexander/git/embox/src/compat/cxx/include -nostdinc -MMD -MP  -D__EMBOX_VERSION__=\"0.4.0\" -include build/base/gen/include/config/embox/arch/x86/kernel/acpi_shutdown.h -D__EMBUILD_MOD__=embox__arch__x86__kernel__acpi_shutdown -c -o build/base/obj/src/arch/x86/kernel/acpi_shutdown.o src/arch/x86/kernel/acpi_shutdown.c
src/arch/x86/kernel/acpi_shutdown.c:10:18: fatal error: acpi.h: No such file or directory
```

ACPI is required to make `shutdown -p` to work.

alexkalmuk, 2020-03-07:
It looks like embox.arch.x86.kernel.acpi_shutdown needs BuildDepends on third_party.lib.acpica at least.

puranjaymohan, 2020-03-09:
Hi @alexkalmuk , I tried adding the `@BuildDepends(third_party.lib.acpica)` in the Mybuild file, and it still didn't work then I added the following two includes 
```
@IncludePath("$(THIRDPARTY_DIR)/lib/acpica")
@IncludePath("$(EXTERNAL_BUILD_DIR)/third_party/lib/acpica/acpica-unix-220150204/source/include/")
```
then it gave an error because the acpi_shutdown.c has a `break;` inside a function on
line no 19, I changed it to a return and built it again. This time it built but when I run it on qemu it crashes.

I am trying to debug it more.

alexkalmuk, 2020-03-09:
@puranjaymohan Ok, thanks. You can probably just change `break` to `return`.

alexkalmuk, 2020-03-09:
> Can you help me with fixing this issue, by explaining why exactly this issue is happening?
> The build is fine after adding the two includes, but when we boot the system, the ACPI initialization fails.

How did it fail? Any messages?

puranjaymohan, 2020-03-09:
> > Can you help me with fixing this issue, by explaining why exactly this issue is happening?
> > The build is fine after adding the two includes, but when we boot the system, the ACPI initialization fails.
> 
> How did it fail? Any messages?

after the following changes:-
1.  Adding the includes for file paths.
```
+@BuildDepends(third_party.lib.acpica)
 module acpi_shutdown extends shutdown {
+       @IncludePath("$(THIRDPARTY_DIR)/lib/acpica")
+       @IncludePath("$(EXTERNAL_BUILD_DIR)/third_party/lib/acpica/acpica-unix-20150204/source/include/")
        source "acpi_shutdown.c"

```
2. Including the `embox.arch.x86.kernel.acpi_shutdown`

I get the following error after running the qemu script

```
	unit: initializing third_party.lib.acpica: EXCEPTION [0xe]: error = 00000000
EAX=0000040e    EBX=002a1670 ECX=00000010 EDX=01bd9ebc
 GS=00150010     FS=00000010  ES=00150010  DS=01bd0010
EDI=002a3000    ESI=001e3fce EBP=01bd9e98 EIP=0017a40b
 CS=00000008 EFLAGS=00010202 ESP=00000050  SS=00000001

 --   00001000 * A R    thread 0  task 1 --------------------------------------

 10 0x00154649 <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
  9 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
  8 0x0017a40b <AcpiFindRootPointer+0x5f> ...e/components/tables/tbxfroot.c:232
  7 0x0017a6bf <AcpiInitializeTables+0x9e> ...e/components/tables/tbxface.c:179
  6 0x00154a51 <__module_third_party__lib__acpica_text_vma+0x45> ...a/init.c:15
  5 0x00153da0 <unit_mod_enable+0x65>                   src/framework/unit.c:25
  4 0x00154272 <mod_enable+0x89>                   src/framework/mod/core.c:130
  3 0x00153c78 <runlevel_set+0xbd>                  src/framework/runlevel.c:54
  2 0x00101dad <init+0x36>                                 src/kernel/init.c:53
  1 0x00101d43 <kernel_start+0x1c>                         src/kernel/init.c:23

run 0x0018bf08 <boot_stub+0x0>                src/kernel/sched/boot_thread.c:20

 --   00001000     R    thread 1  task 1 --------------------------------------

  8 0x0010ccf6 <thread_context_switch+0x50> ...kernel/thread/thread_switch.c:37
  7 0x0018b8bb <__schedule+0x10c>                  src/kernel/sched/sched.c:369
  6 0x0018b94a <sched_preempt+0x1f>                src/kernel/sched/sched.c:426
  5 0x0010a5a5 <critical_dispatch_pending+0x6e>        src/kernel/critical.c:21
  4 0x00100c86 <irq_handler+0x93>          src/arch/x86/kernel/irq_handler.c:16
  3 0x0010004c <_traps_text_start+0x1c>      src/arch/x86/kernel/irq_entry.S:34
  2 0x00101e77 <arch_idle+0xe>                    src/arch/x86/kernel/arch.c:37
  1 0x0010c04b <thread_trampoline+0xa8>             src/kernel/thread/core.c:69

run 0x0018bd22 <idle_run+0x0>                 src/kernel/sched/idle_thread.c:17

```

alexkalmuk, 2020-03-10:
The problem is here:

```
10 0x00154649 <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
  9 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
  8 0x0017a40b <AcpiFindRootPointer+0x5f> ...e/components/tables/tbxfroot.c:232
```

I would suggest you set a breakpoint in gdb to `AcpiFindRootPointer` and examine why it is failed.

puranjaymohan, 2020-03-12:
@alexkalmuk  I tried to put a breakpoint at `AcpiFindRootPointer` and saw, I found the following
```
234	{
(gdb) n
245	    TablePtr = AcpiOsMapMemory (
(gdb) n
248	    if (!TablePtr)
(gdb) n
257	    ACPI_MOVE_16_TO_32 (&PhysicalAddress, TablePtr);
(gdb) n
excep_stub () at src/arch/x86/kernel/exception_entry.S:49
49		SAVE_AL
```
I think the problem is with `ACPI_MOVE_16_TO_32 (&PhysicalAddress, TablePtr);`
here the address I not what it should be. Please tell me how to proceed as I am not well versed with ACPI.

alexkalmuk, 2020-03-12:
@puranjaymohan Can you please PR where ACPI is fixed? Because now it seems the issue can be not so easy.

alexkalmuk, 2020-03-12:
I mean only ACPICA build.

puranjaymohan, 2020-03-12:
@alexkalmuk I have created a pull request which solves this issue.
Thanks

alexkalmuk, 2020-03-13:
@puranjaymohan Thank you! I created another issue https://github.com/embox/embox/issues/1835 (but it's not required to start working on it if it's too difficult:))

alexkalmuk, 2020-03-13:
Closed via https://github.com/embox/embox/pull/1834

closed 2020-03-13 by alexkalmuk