embox/embox #1805: Improve pthread_mutexattr_xxx

anton-bondarev, 2020-03-09:
Add implementation functions which has not been implemented yet
```
extern int   pthread_mutexattr_destroy(pthread_mutexattr_t *);
//extern int   pthread_mutexattr_getprioceiling(const pthread_mutexattr_t *, int *);
//extern int   pthread_mutexattr_getprotocol(const pthread_mutexattr_t *, int *);
//extern int   pthread_mutexattr_getpshared(const pthread_mutexattr_t *, int *);
extern int   pthread_mutexattr_gettype(const pthread_mutexattr_t *, int *);
extern int   pthread_mutexattr_init(pthread_mutexattr_t *);
//extern int   pthread_mutexattr_setprioceiling(pthread_mutexattr_t *, int);
//extern int   pthread_mutexattr_setprotocol(pthread_mutexattr_t *, int);
//extern int   pthread_mutexattr_setpshared(pthread_mutexattr_t *, int);
extern int   pthread_mutexattr_settype(pthread_mutexattr_t *, int);
```

MrSquanchee, 2020-03-09:
Hello @anton-bondarev , I have done some researches on pthread in linux.
I would like to work on this issue.
Thanks,
Suraj.

anton-bondarev, 2020-03-09:
@MrSquanchee 
I've assigned it with you.
But you should finish your opened PR before starting :)

MrSquanchee, 2020-03-12:
@anton-bondarev . I tried so say that I will solve other pthread issues AS WELL (if time permits).
I didn't say I would leave this lol.
Thanks ,
Suraj.

anton-bondarev, 2020-03-12:
> @anton-bondarev . I tried so say that I will solve other pthread issues AS WELL (if time permits).
> I didn't say I would leave this lol.
> Thanks ,
> Suraj.

Ok, I see :)
There are a couple of pthread issues ;) You can start with any of them, but it looks similar only at first sight. I've assigned it again ;)