embox/embox #1816: Implement atexit()

anton-bondarev, 2020-03-10:
atexit() has not been implemented yet

sk2sat, 2020-03-11:
I would like work on this issue.

anton-bondarev, 2020-03-11:
@sk2sat I've assigned it with you.

sk2sat, 2020-03-11:
@anton-bondarev 
thx.
I have a question. Is the application entry point the `main`? Not `_start` or `__libc_start_main`?
If so, I think apps using `atexit()` need to expecitly call `exit()`.

anton-bondarev, 2020-03-12:
> @anton-bondarev
> thx.
> I have a question. Is the application entry point the `main`? Not `_start` or `__libc_start_main`?
> If so, I think apps using `atexit()` need to expecitly call `exit()`.

@sk2sat 
You should go deeper :).
You have to understand how applications execute in Embox (the similar there is in all systems).  It's not usual loader as Linux where there is start code and system loader which you not available, in Embox you can debug all system parts. I think You should try to set a breakpoint at the start of `ifconfig` command ( for example) and look on backtrace.

@ThePiGrepper open PR #1824, can you try work together? It's rather difficult but interesting system task.

anton-bondarev, 2020-05-01:
@sk2sat @ThePiGrepper Hi, I've enabled 'atexit' task resource in #1967 Now you can use it and  #1824 for solving the issue

ThePiGrepper, 2020-05-01:
> @sk2sat @ThePiGrepper Hi, I've enabled 'atexit' task resource in #1967 Now you can use it and #1824 for solving the issue

Nice, Ill take a look at the new code.

anton-bondarev, 2020-06-29:
Closed by #1824

anton-bondarev, 2020-06-29:
Closed by #1824

closed 2020-06-29 by anton-bondarev