embox/embox #1818: implement mkfifo()

anton-bondarev, 2020-03-10:
mkfifo() has not been implemented yet

zhuanyanchengkong, 2020-03-11:
> mkfifo() has not been implemented yet

Hi, @anton-bondarev Let me try to solve this problem.

anton-bondarev, 2020-03-11:
@zhuanyanchengkong I've assigned it with you.
But I think it's rather difficult issue!

anton-bondarev, 2020-03-11:
@zhuanyanchengkong Please, let finish #1806 first!

zhuanyanchengkong, 2020-03-25:
@anton-bondarev Hi,I am using mknod to complete the mkfifo function, do you have any suggestions.

anton-bondarev, 2020-03-26:
@zhuanyanchengkong You can try use mknod, of cause, but except inode creation you have to assign it with fifo functionality.
I think It's better to use fs routines directly.
inode_new(), node_alloc() etc.

zhuanyanchengkong, 2020-03-26:
@anton-bondarev  Yes, I also found that mknod command in embox is incomplete. I want to complete the `vfs_mknod ()` function, and if the embox needs to complete the mknod command, it can also be used directly.

zhuanyanchengkong, 2020-04-04:
@anton-bondarev Hi, I have some problems. The current mknod command cannot set the device number, so do I only need to set the mode in `child = __vfs_subtree_create_child(*tmp_parent, path, len, mode);` to S_IFIFO to use the current mknod code to implement mkfifo?