embox/embox #1819: Implement shm_open() & shm_unlink()

anton-bondarev, 2020-03-10:
shm_open() & shm_unlink() have not been implemented yet

ameetsaahu, 2020-03-13:
Hey, @anton-bondarev I would like to work on this issue.
Can you please help me with how to do it or provide me some resources.

anton-bondarev, 2020-03-14:
@ameetsaahu Ok, let's try. It's rather difficult issue.

I think You have to create some unit test at first.
You can find some example for shm_open in google. [For example ](https://gist.github.com/garcia556/8231e844a90457c99cc72e5add8388e4).

When shared memory object is been opening (creating) it must save some information about itself in some repo. And when open again in another task (process) the information may be found in the repo.

ameetsaahu, 2020-03-15:
Sure, I will start on it today.

ameetsaahu, 2020-03-15:
@anton-bondarev How can I create a unit test for this? Also is there any other implementation already done in embox that is similar to this? I can get some ideas about how I'm gonna try this.

anton-bondarev, 2020-03-15:
@ameetsaahu You can look at scr/tests/posix to know how to create a unit test. I think you should start with it.

PR #2112 (2020-08-13): compat: partially implement shared memory API
Close #1819
closed 2020-08-13 by anton-bondarev