embox/embox #1821: Add tests for pthread_cancel

anton-bondarev, 2020-03-10:
Add test cases for 
```pthread_cancel()
pthread_cleanup_push()
pthread_cleanup_pop()
...
```

sobhaniiest, 2020-03-18:
Hi, I am working on this issue and am trying with a simple test case to test `pthread_cancel`, the code snippet is the following: 

> static pthread_t tid;
static void *thread_fun(void *arg) {
    int r = pthread_cancel(tid);
    test_assert_equal(r, 0);
    return NULL;
}
TEST_CASE("Cancel self thread") {
    void *res;
    test_assert_zero(pthread_create(&tid, NULL, thread_fun, NULL));
    test_assert_zero(pthread_join(tid, &res));
}

The test falls in an infinite loop with the repetition of this:

> run 0x001599bf <boot_stub+0x0>                src/kernel/sched/boot_thread.c:20
EXCEPTION [0xe]: error = 00000000
EAX=a5a5a5a5    EBX=00258620 ECX=00000000 EDX=a5a5a5a5
 GS=00000010     FS=00100010  ES=001a0010  DS=00100010
EDI=00000000    ESI=0025a428 EBP=01b8e600 EIP=0010279d
 CS=00000008 EFLAGS=00010046 ESP=01b8e630  SS=00258620
 --   00001000 * A R    thread 0  task 1 --------------------------------------
 62 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 61 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 60 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 59 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 58 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 57 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 56 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 55 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 54 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 53 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 52 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 51 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 50 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 49 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 48 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 47 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 46 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 45 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 44 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 43 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 42 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 41 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 40 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 39 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 38 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 37 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 36 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 35 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 34 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 33 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 32 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 31 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 30 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 29 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 28 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 27 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 26 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 25 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 24 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 23 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 22 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 21 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 20 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
 19 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
 18 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
 17 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 16 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
[... 9 lines cut ...]
  6 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
  5 0x0010279d <stack_iter_next+0x66>    src/arch/x86/lib/debug/stack_iter.c:40
  4 0x00101bc5 <do_traceback_dump_thread+0x5b>     src/lib/debug/whereami.c:135
  3 0x00101d6c <whereami+0x96>                     src/lib/debug/whereami.c:170
  2 0x0015568c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
  1 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49

I have debugged to get the return value of `pthread_cancel` and got this:

> 21		int r = pthread_cancel(tid);
(gdb) s
pthread_cancel (thread=0x9ccb480 <__pool_storagethread_pool+262144>) at src/compat/posix/pthread/pthread_cancel.c:13
13		int r = thread_cancel(thread);
(gdb) s
[... cut ...]
(gdb) print r
$1 = -201

I have also tried by enabling cancelation state by calling `pthread_setcancelstate(PTHREAD_CANCEL_ENABLE, NULL)`, but got the same result. If I am not wrong, the `pthread_cancel` should call [this](https://github.com/embox/embox/blob/37bbb1e10d4d12e017015b2a8c7a70c34fc69c40/src/kernel/thread/thread_cancel_enable.c#L15) rather it is calling [this](https://github.com/embox/embox/blob/37bbb1e10d4d12e017015b2a8c7a70c34fc69c40/src/kernel/thread/thread_cancel_disable.h#L18) .

alexkalmuk, 2020-03-19:
@sobhaniiest

> I have also tried by enabling cancelation state by calling pthread_setcancelstate(PTHREAD_CANCEL_ENABLE, NULL), but got the same result. If I am not wrong, the pthread_cancel should call this but it is calling this .

Yes, you can add `embox.kernel.thread.thread_cancel_enable` in your mods.config, then rebuild and check once more in gdb.

sobhaniiest, 2020-03-19:
Yes, now in gdb `pthread_cancel` returns 0 i.e successful termination:

> Breakpoint 1, thread_fun (arg=0x0) at src/tests/posix/pthread/pthread_cancel_test.c:19
19	static void *thread_fun(void *arg) {
(gdb) n
20		int r = pthread_cancel(tid);
(gdb) s
pthread_cancel (thread=0x9ccb4a0 <__pool_storagethread_pool+262144>) at src/compat/posix/pthread/pthread_cancel.c:13
13		int r = thread_cancel(thread);
(gdb) s
thread_cancel (t=0x9ccb4a0 <__pool_storagethread_pool+262144>) at src/kernel/thread/thread_cancel_enable.c:16
16		sched_lock();
(gdb) n
18			for (; t->cleanups.counter > 0; t->cleanups.counter--) {
(gdb) n
26			thread_terminate(t);
(gdb) n
28		sched_unlock();
(gdb) n
30		return ENOERR;
(gdb) n
31	}
(gdb) n
pthread_cancel (thread=0x9ccb4a0 <__pool_storagethread_pool+262144>) at src/compat/posix/pthread/pthread_cancel.c:14
14		return r;
(gdb) print r
$1 = 0

But when I test the `pthread_cancel_test` by adding 

> include embox.kernel.thread.thread_cancel_enable
include embox.compat.posix.pthread
@Runlevel(2) include embox.test.posix.pthread.pthread_cancel_test

in my` mods.config`, it again fails and gives following:

> test: running embox.test.posix.pthread.pthread_cancel_test EXCEPTION [0xe]: error = 00000000
EAX=00000000    EBX=002586d0 ECX=00000000 EDX=00001000
 GS=00100010     FS=09ce0010  ES=00100010  DS=00250010
EDI=00000000    ESI=00000000 EBP=09ceb3c8 EIP=0010d6a3
 CS=00000008 EFLAGS=00010202 ESP=53535353  SS=53535353
 --   00001000 * A R    thread 4  task 1 --------------------------------------
  5 0x00155a8c <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
  4 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
  3 0x0010d6a3 <thread_cancel+0x64> src/kernel/thread/thread_cancel_enable.c:15
  2 0x00121b55 <thread_fun+0x21> ...ests/posix/pthread/pthread_cancel_test.c:19
  1 0x0010c194 <thread_trampoline+0xa8>             src/kernel/thread/core.c:69
run 0x00121b34 <thread_fun+0x0> ...tests/posix/pthread/pthread_cancel_test.c:19
 --   00001000     R    thread 0  task 1 --------------------------------------
 16 0x0010ce3f <thread_context_switch+0x50> ...kernel/thread/thread_switch.c:37
 15 0x00159770 <__schedule+0x10c>                  src/kernel/sched/sched.c:369
 14 0x001597ff <sched_preempt+0x1f>                src/kernel/sched/sched.c:426
 13 0x0010a703 <critical_dispatch_pending+0x6e>        src/kernel/critical.c:21
 12 0x0010bed6 <sched_unlock+0x24>     src/include/kernel/sched/sched_lock.h:34
 11 0x0010ca49 <thread_launch+0x54>                src/kernel/thread/core.c:372
 10 0x0017a174 <pthread_create+0x108>    src/compat/posix/pthread/pthread.c:129
  9 0x00121bb2 <__test_case_at_line_25+0x30> ...thread/pthread_cancel_test.c:25
  8 0x0017c758 <test_run+0x59>                     src/framework/test/run.c:120
  7 0x0017c6a3 <test_case_run+0x56>                 src/framework/test/run.c:97
  6 0x0017c5c9 <test_suite_run+0xc3>                src/framework/test/run.c:56
  5 0x0017c2ae <test_mod_enable+0x3c>          src/framework/test/registry.c:31
  4 0x00155541 <mod_enable+0x89>                   src/framework/mod/core.c:130
  3 0x00154f41 <runlevel_set+0xbd>                  src/framework/runlevel.c:54
  2 0x0010143d <init+0x36>                                 src/kernel/init.c:53
  1 0x001013d3 <kernel_start+0x1c>                         src/kernel/init.c:23
run 0x00159dbf <boot_stub+0x0>                src/kernel/sched/boot_thread.c:20
 --   00001000     R    thread 1  task 1 --------------------------------------
  8 0x0010ce3f <thread_context_switch+0x50> ...kernel/thread/thread_switch.c:37
  7 0x00159770 <__schedule+0x10c>                  src/kernel/sched/sched.c:369
  6 0x001597ff <sched_preempt+0x1f>                src/kernel/sched/sched.c:426
  5 0x0010a703 <critical_dispatch_pending+0x6e>        src/kernel/critical.c:21
  4 0x0010030a <irq_handler+0x93>          src/arch/x86/kernel/irq_handler.c:16
  3 0x0010004c <_traps_text_start+0x1c>      src/arch/x86/kernel/irq_entry.S:34
  2 0x00101ebf <arch_idle+0xe>                    src/arch/x86/kernel/arch.c:37
  1 0x0010c194 <thread_trampoline+0xa8>             src/kernel/thread/core.c:69

According to [POSIX standards](https://pubs.opengroup.org/onlinepubs/009695399/functions/pthread_setcancelstate.html): The cancelability state and type of any newly created threads, including the thread in which main() was first invoked, shall be `PTHREAD_CANCEL_ENABLE` and `PTHREAD_CANCEL_DEFERRED` respectively. So the cancelation state should be enabled initially, right?

alexkalmuk, 2020-03-20:
@sobhaniiest I was thinking that `phtread_cancel()` should work without pthread_setcancelstate.

alexkalmuk, 2020-03-20:
So if it crashes then probably it's a bug in pthread_cancel() implementation in Embox. You can also check your code on Linux.

alexkalmuk, 2020-03-20:
@sobhaniiest What I'd propose you to do:

* What we know? pthread_cancel should work even without pthread_setcancelstate called before, it should work every time, that is if there are no cancellation handlers then thread just terminates.
* You can make sure twice running the very similar code on Linux and see how it behaves there.
* Since we know your test is correct and if Linux runs it fine you can create an issue describing the bug.
* You can add another test that cancels not the current thread and check if it works.

sobhaniiest, 2020-03-20:
Hi @alexkalmuk, I have written  the following code and have run in my Linux system:

> #include <stdio.h>
#include <unistd.h>
#include <pthread.h>
static pthread_t target_tid;
/* Cancel self thread */
static void *thread_fun (void *arg)
{
    if (0 == pthread_cancel (pthread_self ()))
        printf ("Thread 1 cancel returns 0\n");
    printf ("This line will not be printed in thread_fun\n");
    return NULL;
}
/* Cancel other thread */
static void *main_thread (void *arg)
{
    sleep(2);
    if (0 == pthread_cancel (target_tid))
        printf ("Thread 2 cancel returns 0\n");
    printf ("This line will be printed in main_thread\n");
    return NULL;
}
static void *target_thread (void *arg)
{
    printf("Target thread is running\n");
    sleep(4);
    printf ("This line will not be printed in target_thread\n");
    return NULL;
}
int main (int argc, char *argv[])
{
    pthread_t tid1, tid2, tid3;
    void *res;
    if (0 == pthread_create (&tid1, NULL, thread_fun, NULL))
        printf ("Thread 1 created\n");
    if (0 == pthread_join (tid1, &res))
    {
        printf ("Thread 1 joined\n");
        if (res == PTHREAD_CANCELED)
            printf ("Thread 1 canceled\n\n");
    }
    if (0 == pthread_create (&tid2, NULL, main_thread, NULL))
        printf ("Thread 2 created\n");
    if (0 == pthread_create (&target_tid, NULL, target_thread, NULL))
        printf ("Thread 3 created\n");
    if (0 == pthread_join (tid2, &res))
        printf ("Thread 2 joined\n");
    if (0 == pthread_join (target_tid, &res))
    {
        printf ("Thread 3 joined\n");
        if (res == PTHREAD_CANCELED)
            printf ("Thread 3 canceled\n");
    }
    return 0;
}

Here I have taken two cases:
1. Thread cancels itself. 
2. One thread cancels another.
Here is the output of the above code:

>  Thread 1 created
Thread 1 cancel returns 0
Thread 1 joined
Thread 1 canceled
Thread 2 created
Thread 3 created
Target thread is running
Thread 2 cancel returns 0
This line will be printed in main_thread
Thread 2 joined
Thread 3 joined
Thread 3 canceled

`pthread_cancel `works as expected in Linux without setting `pthread_setcancelstate` beforehand.

sobhaniiest, 2020-03-20:
May I create an issue?

anton-bondarev, 2020-03-20:
> Can I create an issue?

@sobhaniiest yes, sure! :)