embox/embox #1833: removing getopt_init()

ThePiGrepper, 2020-03-12:
Apparently getopt_init() is not needed anymore, it would be a good idea to start removing it altogether.

ThePiGrepper, 2020-03-12:
apparently getopt() tests failed when removing getopt_init(). are you sure getopt_init() isn't needed anymore?

alexkalmuk, 2020-03-13:
> apparently getopt() tests failed when removing getopt_init(). are you sure getopt_init() isn't needed anymore?

`getopt_init()` is required. It doesn't required for _commnads only_ because it is already called before each command execution - `src/framework/cmd/core.c:33`

ThePiGrepper, 2020-03-13:
I see. I was wondering about that. thanks for the confirmation. I'll test it and send the patch.

alexkalmuk, 2020-03-18:
@ThePiGrepper Thank you!

This issue continues here https://github.com/embox/embox/issues/1863

closed 2020-03-18 by alexkalmuk