embox/embox #1835: acpica is broken

alexkalmuk, 2020-03-13:
Template: `x86/qemu`

Add `include embox.arch.x86.kernel.acpi_shutdown` and make.

```
	unit: initializing embox.fs.buffer_cache: done
	unit: initializing embox.kernel.time.kernel_time: done
	unit: initializing embox.driver.usb.core: done
	unit: initializing embox.driver.usb.class.mass_storage: done
	unit: initializing embox.driver.ide: done
	test: running embox.test.math.math_test .. done
runlevel: init level is 1
	unit: initializing third_party.lib.acpica: EXCEPTION [0xe]: error = 00000000
EAX=0000040e    EBX=002a2590 ECX=00000010 EDX=01bdaebc
 GS=00000010     FS=00000010  ES=00000010  DS=00000010
EDI=002a4000    ESI=001e4e8e EBP=01bdae98 EIP=0017af9c
 CS=00000008 EFLAGS=00000202 ESP=00000050  SS=0017d2f3

 --   00001000 * A R    thread 0  task 1 --------------------------------------

 10 0x00155331 <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
  9 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
  8 0x0017af9c <AcpiFindRootPointer+0x5f> ...e/components/tables/tbxfroot.c:232
  7 0x0017b258 <AcpiInitializeTables+0x9e> ...e/components/tables/tbxface.c:179
  6 0x0015573b <__module_third_party__lib__acpica_text_vma+0x45> ...a/init.c:15
  5 0x00154a82 <unit_mod_enable+0x65>                   src/framework/unit.c:25
  4 0x00154f5a <mod_enable+0x89>                   src/framework/mod/core.c:130
  3 0x0015495a <runlevel_set+0xbd>                  src/framework/runlevel.c:54
  2 0x00101df5 <init+0x36>                                 src/kernel/init.c:53
  1 0x00101d8b <kernel_start+0x1c>                         src/kernel/init.c:23

run 0x0018caee <boot_stub+0x0>                src/kernel/sched/boot_thread.c:20
```

puranjaymohan, 2020-03-15:
I tried putting a breakpoint on `AcpiFindRootPointer` and saw that the address returned by the `AcpiOsMapMemory` is not accessible, it might be because the implementation is wrong.  

```
Breakpoint 1, AcpiFindRootPointer (TableAddress=0x1bdbebc)
    at build/base/gen/../../extbld/third_party/lib/acpica/acpica-unix-20150204/source/components/tables/tbxfroot.c:234
234	{
(gdb) n
245	    TablePtr = AcpiOsMapMemory (
(gdb) n
248	    if (!TablePtr)
(gdb) print TablePtr 
$1 = (UINT8 *) 0x40e <error: Cannot access memory at address 0x40e>
(gdb) print *TablePtr 
Cannot access memory at address 0x40e
```

alexkalmuk, 2020-03-15:
```
void *AcpiOsMapMemory(ACPI_PHYSICAL_ADDRESS Where, ACPI_SIZE Length)
```

And what is `Where` value?

sobhaniiest, 2020-03-15:
Yup, I have also got the same and the `Where` value is `1038`
`AcpiOsMapMemory (Where=1038, Length=2)
at third-party/lib/acpica/osemboxxf.c:103
`

alexkalmuk, 2020-03-16:
@sobhaniiest Yes, thanks.

@puranjaymohan

```
245	    TablePtr = AcpiOsMapMemory (
246	                (ACPI_PHYSICAL_ADDRESS) ACPI_EBDA_PTR_LOCATION,
247	                ACPI_EBDA_PTR_LENGTH);

```

I looks like we should map this memory first. For instance, it could be done with `mmap_device_memory`. You can find some examples with "grep -nr mmap_device_memory src/drivers/"

Most likely it can be done like:

```
mmap_device_memory((void *) ACPI_EBDA_PTR_LOCATION,
                ACPI_EBDA_PTR_LENGTH,
                PROT_READ | PROT_WRITE | PROT_NOCACHE,
                MAP_FIXED,
                ACPI_EBDA_PTR_LOCATION));
```

puranjaymohan, 2020-03-16:
I have added pull request #1876 which adds the call to `mmap_device_memory`.
I am not sure how should the `-EIO` be returned from the function of type void*. I have cast it to void*, don't know if it is correct.

alexkalmuk, 2020-03-17:
> Should a NULL be returned when the error occurs instead of -EIO?

Yes, for sure.

alexkalmuk, 2020-03-17:
@puranjaymohan Thanks for your contribution! I think the issue is completed.

closed 2020-03-17 by alexkalmuk