embox/embox #1837: Arch-independent `cpuinfo` command

Deryugin, 2020-03-13:
Right now `cpuinfo` commands just prints content of this structure, which is x86-specific.

```
struct cpu_info {
	char vendor_id[13];
	unsigned int base_freq;
	unsigned int max_freq;
	unsigned int FPU;
	unsigned int SSE;
	unsigned int MMX;
};
```

It would be nice to extend it somehow to work with other architectures as well.

I guess it's ok to have some generic fields like `vendor_name[]` and `cpu_frequency`, but for extensions we need to rewrite this utility to use arch-agnostic interface.

May be something like this would work

```
struct cpu_feature {
    char name[];
    bool is_supported;
}

int get_cpu_features(struct cpu_feature **features_array) {
    /* Returns number of known features for this arch
        and fills array with corresponding info */
}
```

checker5965, 2020-03-15:
Hi @Deryugin - I can take a look at this issue. Could you please assign it to me?

Just to be clear, we want to make the `cpuinfo` command arch independent to solve this issue. Do we also need to implement the functionality for other architectures apart from x86?

anton-bondarev, 2020-03-16:
> Hi @Deryugin - I can take a look at this issue. Could you please assign it to me?

@checker5965 I've assigned it

> Just to be clear, we want to make the `cpuinfo` command arch independent to solve this issue. Do we also need to implement the functionality for other architectures apart from x86?

Yes, we want to make `cpuinfo` arch-independent. It's useful to enable it on some arm template

Deryugin, 2020-03-16:
@checker5965 
> Do we also need to implement the functionality for other architectures apart from x86?
Sure, it would be nice to detect if FPU/Neon extensions are implemented for given ARM platform, for example. But that's another issue, I guess.

checker5965, 2020-04-10:
@Deryugin Could you please close this issue :)