embox/embox #1840: Implement memccpy()

Deryugin, 2020-03-13:
`memcpy()` is a part of POSIX standard.

For more details refer to POSIX specification: https://pubs.opengroup.org/onlinepubs/9699919799/

 Also implement some tests as well to make sure it works correctly.

tech-gian, 2020-03-13:
Hello. I would like to try solving this issue. I have sent you mails for GSoC from: sdi1900059@di.uoa.gr. If I have a fully understand for this issue, the goal is to implement a function in C, that behaves exactly like memcpy() ? And, also, to implement corresponding tests?

anton-bondarev, 2020-03-14:
> Hello. I would like to try solving this issue. I have sent you mails for GSoC from: [sdi1900059@di.uoa.gr](mailto:sdi1900059@di.uoa.gr). If I have a fully understand for this issue, the goal is to implement a function in C, that behaves exactly like memcpy() ? And, also, to implement corresponding tests?

Yes, you got in right!

I've assigned it for you

tech-gian, 2020-03-14:
Thank you for the assignment.
I want to ask one thing. Original memcpy() doesn't handle overlap, instead it runs the same process (copies each byte) like in any other situation. Do you want me to do something different or to keep the same plan?

alexkalmuk, 2020-03-15:
@tech-gian `memcpy` is already implemented, the issues tells about `memccpy`:

```
void *memccpy(void *dest, const void *src, int c, size_t n);
```

tech-gian, 2020-03-15:
Oh, sorry. I missed that c. The question remains, how to handle overlap in memccpy()? Just like any other case, probably with undefined results?

alexkalmuk, 2020-03-15:
POSIX claims that:

```
If copying takes place between objects that overlap, the behavior is undefined.
```

So yes, the result will be undefined.

tech-gian, 2020-03-15:
Ok. Thank you for the clarification

tech-gian, 2020-03-16:
I have finished (almost) my memccpy() function and I have almost finished all of the tests that need to be passed for the correctness of the function. I think that today or tomorrow, everything will be ready. I was working it in a private repository. What do I need to do so I can make it through embox's public repository? Just a pull request of the implementation and the tests? Is there any guide that you suggest, to understand what to do? (I'm sorry that I'm asking so basic things but I've been using github only for a couple of months and I'm not yet familiar with all its functionality)

anton-bondarev, 2020-03-16:
@tech-gian Yes, now you should create a new PR which will be reviewed and approved one of the maintainers. Then your PR will be merged to embox/master.

Your PR should contain commits with tests and memccpy() implementation.

Let's try to create the first PR :)

Deryugin, 2020-03-16:
@tech-gian 
> Is there any guide that you suggest, to understand what to do?

You can check some pages on our wiki:
https://github.com/embox/embox/wiki/Contributing%3A-Git-workflow
https://github.com/embox/embox/wiki/Code-Style

tech-gian, 2020-03-16:
Thank you very much! I really appreciate your help:) I'm finishing with some details (comments etc) and I hope it will be finished in a few hours.

tech-gian, 2020-03-16:
I have created a PR (or at least I tried my best). Ι think that i linked that issue with that PR. I am sorry that there is only one commit, but I was working this issue in a local folder. It won't happen again, now that I understood how everything works. Please let me know for any problems about the implementation, tests or even the PR. I want to learn, even from my mistakes!

PR #1894 (2020-03-27): 1840 memccpy implementation
closes #1840 (Good First issue)
After correcting my mistake in the first PR, I made this try and now I hope everything is ok.
I have fixed memccpy_test.c, renamed memccpy.c, so everyone can use it, and I think I have connected these with Mybuild correctly.
I ran embox emulator and it didn't get me any error. Please let me know if something is wrong or need to be fixed.

Thank you for your time
closed 2020-03-27 by anton-bondarev