embox/embox #1845: Add tests for ffs()

anton-bondarev, 2020-03-14:
Implement some test cases for standard [ffs()](https://pubs.opengroup.org/onlinepubs/009695399/functions/ffs.html) routine

binkoni, 2020-03-14:
Hello sir, can I assigned to this issue?
I want to participate this project as a GSOC.
I didn't send an email for contact yet, but I'm planning to send the email which contains detailed information about me after solve some issues here

anton-bondarev, 2020-03-14:
@binkoni I've assigned it to you.
But you should send email to any of mentors at first. It can be short information, and then you can provide more details.

binkoni, 2020-03-14:
@anton-bondarev Okay I'll compose the email right now.

binkoni, 2020-03-14:
@anton-bondarev Can I ask a basic question
I'm wondering how to run unit tests.
I added an unit test from other pull request to conf/mods.config such as:
```console
@Runlevel(2) include embox.test.posix.pthread_rwlock
```
And when I run make, build fails like this:
```console
 CONFIGFILE conf/mods.config                                                                                                                                                                                                                 
 CONFIGLINK                                                                                                                                                                                                                                  
conf/mods.config:93:23: error: Couldn't resolve reference to ModuleType 'embox.test.posix.pthread_rwlock'.                                                                                                                                   
 BUILDMODEL                                                                                                                                                                                                                                  
 BUILDGEN build                                                                                                                                                                                                                              
Error: modules dependency cycle:                                                                                                                                                                                                             
        embox.fs.driver.devfs_old                                                                                                                                                                                                            
        embox.fs.rootfs                                                                                                                                                                                                                      
        embox.fs.syslib.file_system_full                                                                                                                                                                                                     
        embox.driver.common                                                                                                                                                                                                                  
Error: modules dependency cycle:                                                                                                                                                                                                             
        embox.fs.driver.devfs_old                                                                                                                                                                                                            
        embox.fs.rootfs                                                                                                                                                                                                                      
        embox.fs.syslib.file_system_full                                                                                                                                                                                                     
        embox.driver.common                                                                                                                                                                                                                  
```

anton-bondarev, 2020-03-15:
@binkoni 

`@Runlevel(2) include embox.test.posix.pthread_rwlock` You use the wrong module full name.

You should use `embox.test.posix.pthread.pthread_rwlock` because for  this module `package embox.test.posix.pthread` and name is pthread_rwlock

> Also added unit test fails on boot.

Which tests fails?

binkoni, 2020-03-15:
Oh sorry, pthread_rwlock test was my mistake, it runs fine.
I also tried math_test_fp yesterday by adding 
```console
@Runlevel(2) include embox.test.math.math_test_fp
```
And I thought that I was running unit tests in a wrong way because result was like below:
```console
        test: running embox.test.math.math_test_fp .EXCEPTION [0xe]: error = 00000000                    
EAX=00259fa0    EBX=00259fa0 ECX=000001bb EDX=40040000                                                   
 GS=00000010     FS=00000010  ES=00000010  DS=00000010
EDI=0025b000    ESI=001b0fd4 EBP=01a41f04 EIP=00157a57
 CS=00000008 EFLAGS=00010206 ESP=00000000  SS=00000000

 --   00000000 * A   W  thread 1076232192  task 2465696 -----------------------

 62 0x00155685 <exception_handler+0x125> ...h/x86/kernel/exception_handler.c:20
 61 0x00100079 <excep_stub+0x1c>       src/arch/x86/kernel/exception_entry.S:49
 60 0x00157a57 <clock_tick_handler+0x7b>             src/kernel/time/timer.c:29
 59 0x0010a3c9 <irq_dispatch+0x13b>                        src/kernel/irq.c:155
 58 0x00100d25 <irq_handler+0x67>          src/arch/x86/kernel/irq_handler.c:16
 57 0x0010004c <_traps_text_start+0x1c>      src/arch/x86/kernel/irq_entry.S:34
 56 0x0019e35f <__x86.get_pc_thunk.ax+0x0> 
 55 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 54 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 53 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 52 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 51 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 50 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 49 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 48 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 47 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 46 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
 45 0x0014ceff <fmod+0x3c>              src/compat/libc/math/math_builtins.c:16
```

alexkalmuk, 2020-03-15:
math_test_fp requires `math_openlibm` instead of `math_builtins`.

alexkalmuk, 2020-03-16:
@binkoni Thank you!

Did you follow the steps here - https://github.com/embox/embox/wiki/Google-Summer-of-Code-2020#introduction? (I do not see email from you).

closed 2020-03-16 by alexkalmuk

binkoni, 2020-03-16:
@alexkalmuk Yes, I emailed to @Deryugin regarding memory profiling project few days ago.

alexkalmuk, 2020-03-16:
@binkoni Ah, ok then :) He confirmed that he received your email.

binkoni, 2020-03-16:
Okay thanks!

PR #1864 (2020-03-16): compat/libc: Added test code for ffs()
This fixes #1845
This commit adds test code for ffs()
commit 5a38d491 (2020-03-16): compat/libc: Added test code for ffs()

This fixes #1845
This commit adds test code for ffs()