embox/embox #1856: Add unit tests for strerror()

anton-bondarev, 2020-03-14:
Add some test cases for standard strerror()

tanner00, 2020-03-25:
Can I be assigned this?

anton-bondarev, 2020-03-26:
@tanner00 I've assigned
Have you already sent email to any of mentors?

tanner00, 2020-03-27:
Yes

85ayush, 2021-02-20:
Hello I would like to work on this issue, I am new to open source contribution, can you please help me , and I will give my 100%

anton-bondarev, 2021-02-20:
@85ayush Hi, I've assigned it to you
You should start with [Quick-Start](https://github.com/embox/embox/wiki/Quick-Start). First, build and run Embox on QEMU, and then add tests and prepare a new PR with your changes

85ayush, 2021-02-21:
Thank you, I will do my best.

85ayush, 2021-02-21:
I have done the steps upto the orange mark on the image below. I mean I
have 1. Clonned
2.installed docker
But where do I have to use the command "docker -machine Is"

85ayush, 2021-02-21:
Can you please guide me for the issue described in the before male.

85ayush, 2021-02-22:
After using". /scripts/docker/docker_start.sh" Command, at the end it says
can't start container, is it fine or there is some error? and when I use "
make cofload-64/qemu" it says " make: command not found". Can you please
help me with this?

85ayush, 2021-02-22:
It says "autoqemu_arch is empty" After using command
"./script/qemu/auto_qemu", can you please help me what does it mean?

anton-bondarev, 2021-02-23:
@85ayush  Do you use Windows?
I think the simplest way work with Embox on Windows is to install virtual machine (VirtualBox for example) with Linux.
Or you can use WSL2, it works for me

> It says "autoqemu_arch is empty" After using command
> "./script/qemu/auto_qemu"
> 

Have you built Embox?
What template have you used for configuration?

85ayush, 2021-02-24:
issue.mkv
<https://drive.google.com/file/d/12UrQxdZDEmochjEL5S_Obs5pAmPbvs5v/view?usp=drive_web>
here I have given the video of what I am doing to resolve the issue, can
you please tell me where I am making mistake. And can you please guide me
how to do it?

anton-bondarev, 2021-02-28:
@85ayush Hi
First, you have to build a standard Embox template and run it on QEMU

It's enough to do the following commands
```
make confload-x86/qemu
make 
./scripts/qemu/auto_qemu
```
I don't know how I can see where you have done mistakes there are a lot of variants. You must find out in your environment yourself.
Of course, you should ask, when you see trouble when you run the commands. But you have to give more correct descriptions

PR #2636 (2022-12-16): Add some tests for strerror
Fix #1856
closed 2022-12-16 by anton-bondarev