embox/embox #1859: Add unit test for memset()

anton-bondarev, 2020-03-14:
Implement some test cases for standard memset()

binkoni, 2020-03-14:
I would like to get assigned.

alexkalmuk, 2020-03-15:
@binkoni Ok, assigned this one to you.

alexkalmuk, 2020-03-15:
Ah, sorry, let's it's your second "good first issue", but we are required to solve only one. Let's finish with https://github.com/embox/embox/pull/1862

binkoni, 2020-03-15:
Oh I didn't knew about that. thanks

KhaledEmaraDev, 2020-03-25:
Hi @alexkalmuk, couls you assign this issue to me? I have already contacted you by email.

anton-bondarev, 2020-03-25:
@KhaledEmaraDev I've assigned it

Jia1018, 2021-02-11:
Hello im new here and i would like to start contributing.
Is this thread still open/ not being worked on by anyone ?
If so I would also try to get my fingers wet :)
Just let me know if there are anything i can do.

anton-bondarev, 2021-02-11:
@Jia1018 I've assigned it to you

Jia1018, 2021-02-14:
Please excuse me for relying this late. But I noticed that the PR of this issue has already been closed, so i wonder whether you still need me to work on this:)

Jia1018, 2021-02-14:
Or i can help with another issue:)

anton-bondarev, 2021-02-14:
@Jia1018 the issue is not closed. You could work on it on another as you wish

Jia1018, 2021-02-14:
ok, thanks. I'll do my best:)

发自我的iPhone

semtagg, 2023-09-28:
Hi! Is this issue still available? Can I take it?

anton-bondarev, 2023-09-29:
Hello @semtagg I've assigned it to you

PR #2954 (2023-09-30): tests: Add unit test for memset()
Close #1859
closed 2023-09-30 by anton-bondarev