embox/embox #1860: Add unit test for memcmp()

anton-bondarev, 2020-03-14:
Implement some test cases for memcmp()

binkoni, 2020-03-14:
Hello @anton-bondarev , I would like to get assigned.

sha-ahammed, 2020-03-16:
Hello sir,
I would like to do this as part of GSOC.

anton-bondarev, 2020-03-16:
@sha-ahammed Hello
I've assigned it for you, but you should send email to one of the mentors at first.

sha-ahammed, 2020-03-16:
Oh my apologies, i was thinking solve issue first and mail later.Is there a specific mentor i should contact?

anton-bondarev, 2020-03-17:
@sha-ahammed No, there isn't.
All mentors can review your changes and help you if it's required

sha-ahammed, 2020-03-18:
@anton-bondarev hey, how do i run tests?

anton-bondarev, 2020-03-18:
@sha-ahammed Just add to mods.config
You may see tests executing when Embox starts

sha-ahammed, 2020-03-18:
@anton-bondarev I did that and i configured the project on my fork. There was no error aside from 

`Error: modules dependency cycle:
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
	embox.driver.common
Error: modules dependency cycle:
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
	embox.driver.common
`
Does this mean the test was successfull? Can i submit a Pull Request?
Thank you.

sha-ahammed, 2020-03-18:
@anton-bondarev ok so it doesnt seem to be showing up
`runlevel: init level is 0
	unit: initializing embox.mem.static_heap: done
	unit: initializing embox.profiler.tracing: done
	unit: initializing embox.kernel.sched.sched: done
	unit: initializing embox.driver.interrupt.i8259: done
	unit: initializing embox.fs.buffer_cache: done
	unit: initializing embox.driver.clock.pit: done
	unit: initializing embox.kernel.time.kernel_time: done
	unit: initializing embox.kernel.time.timekeeper: done
	unit: initializing embox.fs.rootfs: done
	unit: initializing embox.driver.flash.flash_nofs: done
	unit: initializing embox.kernel.time.timer_handler: done
	unit: initializing embox.driver.usb.core: done
	unit: initializing embox.driver.usb.class.mass_storage: done
	unit: initializing embox.driver.ide: done
	test: running embox.test.math.math_test .. done
runlevel: init level is 1
	unit: initializing embox.net.neighbour: done
	unit: initializing embox.mem.slab: done
	unit: initializing embox.driver.pci: done
	unit: initializing embox.net.net_entry: done
	unit: initializing embox.net.tcp: done
	unit: initializing embox.driver.tty.serial: done
	pci: Apple OHCI usb host driver inserted
	pci: Apple OHCI usb host handles 106b:003f bus 0 slot 4 func 0
	unit: initializing embox.kernel.sched.sched_ticker_preempt: done
	pci: virtio driver inserted
	pci: virtio handles 1af4:1000 bus 0 slot 3 func 0
	unit: initializing embox.driver.net.loopback: done
	unit: initializing embox.driver.serial.i8250: done
	test: running embox.test.math.fpu_context_consistency_test . done
	test: running embox.test.posix.getopt_test .............. done
	test: running embox.test.posix.environ_test ......... done
	test: running embox.test.stdlib.qsort_test ....... done
	test: running embox.test.stdlib.bsearch_test ...... done
	test: running embox.test.posix.ppty_test ... done
	test: running embox.test.posix.pipe_test ... done
	test: running embox.test.posix.select_test ...... done
	test: running embox.test.posix.poll_test ......... done
	test: running embox.test.stdio.printf_test ......... done
	test: running embox.test.recursion . done
	test: running embox.test.critical .. done
`

sha-ahammed, 2020-03-18:
@anton-bondarev this is what i added to mods.config
`@Runlevel(1) include embox.test.stdlib.memcmp_test`

sha-ahammed, 2020-03-18:
I still cant get the test in output.Am i missing something @anton-bondarev ?

PR #1904 (2020-03-27): Add Unit tests for memcmp
Closes #1860
closed 2020-03-27 by anton-bondarev