embox/embox #1865: CI: ping xip.io fails too often

alexkalmuk, 2020-03-16:
`ping 192.168.128.128.xip.io` fails too often for both expect tests and start_script:

https://github.com/embox/embox/runs/505471758#step:10:266 (start_script)
https://github.com/embox/embox/runs/505907968#step:10:208 (expect)

AntonKozlov, 2020-03-16:
Have you looked at the source of the error? I had no chance to look at.

```
ping: no error 192.168.128.128.xip.io
ping: Command returned with code -22: Invalid argument
error: Unknown error code. PLEASE REPORT ISSUE 519!
```

It fails from time to time and the error message is really suspicious (`no error`). It really looks like a problem with DNS resolve, misreported. And I haven't seen the error on Travis, right? It may mean embox issue reproducing only in certain environments

alexkalmuk, 2020-03-16:
No, I haven't look in more detail yet, but I have very similar observations.

alexkalmuk, 2020-03-16:
@AntonKozlov 

> And I haven't seen the error on Travis, right?

https://travis-ci.org/github/embox/embox/jobs/662978056#L9029