embox/embox #1869: x86/qemu fails to start when compiled with -O2

Deryugin, 2020-03-16:
Looks like problem is related to `telnetd` service

1. `make confload-x86/qemu`
2. Replace `-O0` with `-O2` in `conf/build.conf`
3. `make`
4. `./scripts/qemu/auto_qemu`
```
<...>
runlevel: init level is 2
runlevel: init level is 3
runlevel: init level is 4
Default IO device[ttyS0]
>export PWD=/
>export HOME=/
>partition
>netmanager
>service telnetd
Starting service: telnetd
EXCEPTION [0xe]: error = 00000000
EAX=01b5cf88    EBX=6e6c6575 ECX=0000001f EDX=09c56ca5
 GS=00000010     FS=00000010  ES=00000010  DS=00000010
EDI=00000020    ESI=09c56ca4 EBP=09c56ca4 EIP=0012e7d4
 CS=00000008 EFLAGS=00010202 ESP=00221294  SS=00000000
EXCEPTION [0xe]: error = 00000000
EAX=01b5cc64    EBX=09c56ca4 ECX=00221294 EDX=53535353
 GS=00000010     FS=00000010  ES=00000010  DS=00000010
EDI=01b5cc64    ESI=00100030 EBP=00222000 EIP=00101b40
 CS=00000008 EFLAGS=00010016 ESP=00221294  SS=00000000
<...>
```

layman-n-ish, 2020-03-19:
The bug occurs even at the 1st optimization level: `-O1` (`-O2` includes all optimization flags specified by `-O1` **[1]**) and doing a naive binary search on the optimization flags incorporated by it reveals the flag `-fomit-frame-pointer` to be the source of the exception. Ergo, a "hack" to resolve this bug **for now** could be:

1. `make confload-x86/qemu`
2. Replace `-O0` with `-O2` in `conf/build.conf`
3. Append line: `CFLAGS += -fno-omit-frame-pointer` in `conf/build.conf`
4. `make`
5. `./scripts/qemu/auto_qemu`

References:
**[1]** https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html

--- 

Going by the trap number: `0xe`, a **Page Fault** exception is met with, as defined [here](https://github.com/embox/embox/blob/master/src/arch/x86/include/asm/traps.h#L40). Any leads on how to go ahead investigating this?

alexkalmuk, 2020-03-21:
@layman-n-ish Thanks for the observation! I wonder why `-fno-omit-frame-pointer` help (I checked this). We should investigate the issue deeper.

alexkalmuk, 2020-03-21:
Possibly we can examine the stack with gdb to figure out something about the nature of this page fault. So then maybe we can explain why this flag helps.

layman-n-ish, 2020-03-21:
> Possibly we can examine the stack with gdb to figure out something about the nature of this page fault. So then maybe we can explain why this flag helps.

Certainly! I'm on it.