embox/embox #1870: Add support for '%n' format for scanf()

Deryugin, 2020-03-16:
According to POSIX standard https://pubs.opengroup.org/onlinepubs/9699919799/
```
n
    No input is consumed. The application shall ensure that the
corresponding argument is a pointer to the integer into which
shall be written the number of bytes read from the input so far
by this call to the fscanf() functions. Execution of a %n conversion
specification shall not increment the assignment count returned
at the completion of execution of the function. No argument shall
be converted, but one shall be consumed. If the conversion
specification includes an assignment-suppressing character or
a field width, the behavior is undefined.
```

It's required to extend existing `scanf()` implementation with this format and also add corresponding test for `package embox.test.stdio.scanf_test`.

RubyOcelot, 2020-03-18:
Hi, I would like to take this issue, please assigned to me.

Deryugin, 2020-03-18:
@RubyOcelot ok!

RubyOcelot, 2020-03-18:
@Deryugin I've add %n to scanf and test for scanf. but how to run the test and check the correctness? I tried to add it in mod.config but don't work

anton-bondarev, 2020-03-18:
> I tried to add it in mod.config but don't work

The test fails? Or it is not called?

RubyOcelot, 2020-03-18:
> > I tried to add it in mod.config but don't work
> 
> The test fails? Or it is not called?

It's not called, the test list doesn't change.
I add a line in mod.config for x86/qemu:
`@Runlevel(1) include embox.test.stdio.scanf_test`

anton-bondarev, 2020-03-18:
@RubyOcelot Let me see on your module description? (Mybuild file)

closed 2020-03-20 by Deryugin