embox/embox #1886: mips/qemu don't build with OpenLibm

anton-bondarev, 2020-03-19:
If enabele openlibm in mips/qemu
```
	//include embox.compat.libc.math_builtins
	include embox.compat.libc.math_openlibm
```
build fails
```
./build/base/obj/embox.o: In function `print_f':
src/compat/libc/include/ctype.h:(.text+0x5af00): undefined reference to `roundl'
src/compat/libc/include/ctype.h:(.text+0x5afe8): undefined reference to `roundl'
```

puranjaymohan, 2020-03-20:
I want to work on this issue

puranjaymohan, 2020-04-18:
@anton-bondarev Please tell me how to proceed with solving this issue. I am kind of stuck here.
Thanks

Deryugin, 2020-04-20:
@puranjaymohan I think that problem is in `openlibm-0.7.0/src/Make.files`. For some reason `s_roundl.c` (which implements `roundl()`) is compiled only for `i387 amd64 aarch64` architectures. Looks like it can be solved the same way it was solved for arm (take a look at `embox/third-party/lib/openlibm/patch.txt`).

anton-bondarev, 2020-04-20:
@puranjaymohan Have you reproduced the trouble?

I see in openlibm/patch.txt
```
+ifneq ($(filter $(ARCH),arm),)
+$(CUR_SRCS) += s_roundl.c
+endif
+
```

puranjaymohan, 2020-04-20:
Hi @Deryugin Thanks for the insight, I have used the same method to solve it for mips. 

@anton-bondarev Yes I have reproduced the issue and also solved it now after your suggestions. A pull request has been created #1956.
Thanks

closed 2020-04-22 by Deryugin