embox/embox #1900: Qt build broken with gcc 9

alexkalmuk, 2020-03-24:
Template: `x86/qt-fb` (but I think the same issue can be reproduced for x86/qt-vnc and other qt templates).

```
../../include/QtGui/../../src/gui/styles/qstyleoption.h:878:5: note: because �QStyleOptionGraphicsItem� has user-provided �QStyleOptionGraphicsItem::QStyleOptionGraphicsItem(const QStyleOptionGraphicsItem&)�
  878 |     QStyleOptionGraphicsItem(const QStyleOptionGraphicsItem &other) : QStyleOption(Version, Type) { *this = other; }
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
dialogs/qfiledialog.cpp: In member function �QByteArray QFileDialog::saveState() const�:
dialogs/qfiledialog.cpp:426:33: error: �class Ui_QFileDialog� has no member named �treeView�
  426 |     stream << d->qFileDialogUi->treeView->header()->saveState();
      |                                 ^~~~~~~~
dialogs/qfiledialog.cpp: In member function �bool QFileDialog::restoreState(const QByteArray&)�:
dialogs/qfiledialog.cpp:482:49: error: �class Ui_QFileDialog� has no member named �treeView�
  482 |     QHeaderView *headerView = d->qFileDialogUi->treeView->header();
      |                                                 ^~~~~~~~
dialogs/qfiledialog.cpp: In member function �void QFileDialogPrivate::retranslateStrings()�:
dialogs/qfiledialog.cpp:573:46: error: �class Ui_QFileDialog� has no member named �treeView�
  573 |     QList<QAction*> actions = qFileDialogUi->treeView->header()->actions();
      |                                              ^~~~~~~~
dialogs/qfiledialog.cpp: In member function �void QFileDialog::setOptions(QFileDialog::Options)�:
dialogs/qfiledialog.cpp:679:27: error: �class Ui_QFileDialog� has no member named �newFolderButton�
  679 |         d->qFileDialogUi->newFolderButton->setEnabled(!ro);
      |                           ^~~~~~~~~~~~~~~
dialogs/qfiledialog.cpp: In member function �virtual void QFileDialog::setVisible(bool)�:
dialogs/qfiledialog.cpp:756:27: error: �class Ui_QFileDialog� has no member named �fileNameEdit�
  756 |         d->qFileDialogUi->fileNameEdit->setFocus();
      |                           ^~~~~~~~~~~~
```

287mdsahil, 2020-04-09:
I would like to work on this

alexkalmuk, 2020-07-31:
The same problem is referenced in https://github.com/embox/embox/issues/2024

alexkalmuk, 2020-07-31:
After some investigation, it was figured out that file `build/extbld/third_party/qt/core/qt-everywhere-opensource-src-4.8.7/src/gui/.uic/debug-static/ui_qfiledialog.h` is "incorrect". It contains the following filelds:

```cpp
class Ui_QFileDialog
{
public:
    QGridLayout *gridLayout;
    QLabel *label;
    QHBoxLayout *hboxLayout;
    QFileDialogComboBox *fileDialogComboBox;
    QToolButton *toolButton;
    QToolButton *toolButton1;
    QToolButton *toolButton2;
    QToolButton *toolButton3;
    QToolButton *toolButton4;
    QToolButton *toolButton5;
    QSplitter *splitter;
    QSidebar *sidebar;
    QFrame *frame;
    QVBoxLayout *vboxLayout;
    QStackedWidget *stackedWidget;
    QWidget *widget;
    QVBoxLayout *vboxLayout1;
    QFileDialogListView *fileDialogListView;
    QWidget *widget1;
    QVBoxLayout *vboxLayout2;
    QFileDialogTreeView *fileDialogTreeView;
***
```

For example, `fileDialogTreeView` should be `treeView`.

Ok, there is how file `ui_qfiledialog.h` was created:

```
uic/debug-static/ui_qfiledialog.h: dialogs/qfiledialog.ui
    /home/alexander/git/embox/build/extbld/third_party/qt/core/qt-everywhere-opensource-src-4.8.7/bin/uic dialogs/qfiledialog.ui -o .uic/debug-static/ui_qfiledialog.h
```

So bin/uic is a part of Qt and compiled with host g++ during Qt build. After that `ui_qfiledialog.h` is compiled from `qfiledialog.ui` with this `uic` compiler. What has figured out that output depends on g++ version!

alexkalmuk, 2020-07-31:
If you take a look at `class Ui_QFileDialog` above you can notice that variables named after their class names, but not with the names from dialogs/qfiledialog.ui. So it looks like when UIC is compiled with GCC 7, then it parses dialogs/qfiledialog.ui in a "right way" and compilation succeed. But when UIC is compiled with GCC 9, it doesn't use actual variables names from dialogs/qfiledialog.ui , but just generates them from class names.

alexkalmuk, 2020-07-31:
Additionally, an example with UIC compiled with GCC 7 (from file `ui_qfiledialog.h`):

```
    QToolButton *backButton;
    QToolButton *forwardButton;
    QToolButton *toParentButton;
    QToolButton *newFolderButton;
    QToolButton *listModeButton;
    QToolButton *detailModeButton;
```

Then same stuff with UIC compiled with GCC 9:

```
    QToolButton *toolButton;
    QToolButton *toolButton1;
    QToolButton *toolButton2;
    QToolButton *toolButton3;
    QToolButton *toolButton4;
    QToolButton *toolButton5;
```

closed 2020-08-04 by alexkalmuk