embox/embox #1914: /dev/mem compatibility

Deryugin, 2020-03-30:
Some Linux applications access physical addresses via devfs like this

```
    uintptr_t phys_addr; /* Target physical address */
    size_t size; /* Size of memory region */

    m_fd = open("/dev/mem", O_RDWR | O_SYNC);
    virt_addr = mmap(0, size, PROT_READ | PROT_WRITE,
        MAP_SHARED, m_fd, phys_addr);
   ...
```

It would be nice to implement pseudo device `/dev/mem` for devfs to be able to run such programs.

puranjaymohan, 2020-04-27:
I want to work on this issue, please tell me where should I start and what documentation or code I should read.
Thanks

anton-bondarev, 2020-04-28:
@puranjaymohan You can start with the creation of a new virtual device.
You can give src/drivers/virtual/zero.c as an example.

puranjaymohan, 2020-04-29:
@anton-bondarev I have successfully created a new virtual device with the name "mem" and tested it on x86/qemu, /dev/mem is created. Now I need to understand how character drivers work with embox and how to mimic the operation of /dev/mem device. I tried reading the Linux implementation of /dev/mem, but its char drivers(syntax) are completely different from the embox.

Thanks

anton-bondarev, 2020-04-30:
@puranjaymohan Perfect. Now look at `struct idesc_ops` especially `idesc_mmap` field which s required for mmap() from the issue example

`struct idesc_ops` is the structure for implementation of index descriptor operations.

Embox file sybsustem is similar Linux (Minix) but much easier and therefore with restrictions

puranjaymohan, 2020-04-30:
Hi @anton-bondarev I looked at the struct `idesc_ops` at `src/include/fs/idesc.h`,
and also the `idesc_mmap` at `src/fs/syslib/idesc_mmap/idesc_mmap.c`. Is there an example where this is used and also how to test this?

puranjaymohan, 2020-05-01:
Should I use the void *mmap_device_memory() to map the physical address to virtual address? But it requires the virtual address as a parameter. I think there is an option to send NULL as the parameter and then it should get mapped to any virtual address, but this functionality is not implemented as it is in TODO of that function.

anton-bondarev, 2020-05-01:
> Hi @anton-bondarev I looked at the struct `idesc_ops` at `src/include/fs/idesc.h`,
> and also the `idesc_mmap` at `src/fs/syslib/idesc_mmap/idesc_mmap.c`. Is there an example where this is used and also how to test this?

I think the example in issue description is good for test
```
    uintptr_t phys_addr; /* Target physical address */
    size_t size; /* Size of memory region */

    m_fd = open("/dev/mem", O_RDWR | O_SYNC);
    virt_addr = mmap(0, size, PROT_READ | PROT_WRITE,
        MAP_SHARED, m_fd, phys_addr);
```

anton-bondarev, 2020-05-01:
> Should I use the void *mmap_device_memory() to map the physical address to virtual address? But it requires the virtual address as a parameter. I think there is an option to send NULL as the [...]

I think You should start from simple one-to-one mapping, so virtual and physical addresses are equals. I mean You can return physical address now now

puranjaymohan, 2020-05-01:
@anton-bondarev I have added a pull request which tries to solve this, please have a look.
Thanks

PR #1968 (2020-05-03): Drivers: virtual: mem: added virtual device /dev/mem
Add a character driver that implements /dev/mem file and
helps in accessing physical memory using devfs.
closes #1914
commit 350c84fb (2020-05-03): Drivers: virtual: mem: added virtual device /dev/mem

Add a character driver which implements /dev/mem file and
helps in accesing physical memory using devfs.
closes #1914
closed 2020-05-03 by anton-bondarev