embox/embox #1916: POSIX aligned memory allocation

Deryugin, 2020-03-30:
We're missing some functions for aligned memory allocation

```
       #include <stdlib.h>

       int posix_memalign(void **memptr, size_t alignment, size_t size);
       void *aligned_alloc(size_t alignment, size_t size);
       void *valloc(size_t size);

       #include <malloc.h>
       void *pvalloc(size_t size);
```

We already have `memalign()` function, so it can be used to implement some of these functions.

kouamdo, 2020-06-30:
Please assign me that issue ...
I want to know about align memory allocation in embox

anton-bondarev, 2020-06-30:
> Please assign me that issue

@kouamdo done

alexkalmuk, 2020-07-16:
I just want to mention that `malloc.h` (which declares `pvalloc`) is not a part of POSIX. It's due to discussions in the telegram group.

So maybe to add malloc.h to src/compat/linux, for example.

kouamdo, 2020-07-16:
src/compat/libc/include/malloc.h indeed , okay , thank you :)

alexkalmuk, 2020-07-16:
> src

@kouamdo I would even propose `src/compat/linux/include/malloc.h` of something like this. Because the important point here is that malloc.h is _not_ a part of libc, but a part of GNU libc. So it seems to be reasonable to have it in linux-specific folder.