embox/embox #1923: Static light thread initialization seems useful

anton-bondarev, 2020-04-01:
Light threads are often use in kernel. They require unit_init routine, but sometimes it contains the only light thread initialization. For example: net_entry module, or embox.kernel.time.timer_handler.

It may be useful if light thread was declare as global (static) object which does not require runtime initializing.

alexkalmuk, 2020-04-08:
It's partially completed by https://github.com/embox/embox/pull/1933

Nevertheless, when someone call `lthread_launch` the scheduler must be initialized to this time.

KlyukinSA, 2023-07-16:
Hello @anton-bondarev 
Do I have to make scheduler static (work without `EMBOX_UNIT_INIT(sched_start_init);`)?

anton-bondarev, 2023-07-17:
@KlyukinSA Hello,
I'm not sure about this. But it is one of the available way
As I see there are some LTHREAD_DEF using, but those modules where it is used must be sure that when lthread_launch() is called sheduler haalready s been started.