embox/embox #1929: microblaze: Linker relaxation (--relax) segfault

alexkalmuk, 2020-04-05:
Template: `microblaze/qemu`

Remove `LD_DISABLE_RELAXATION = y` from build.conf and build embox. You will get:
```
-Ibuild/base/gen \
-imacros build/base/gen/config.lds.h \
	-MMD -MT build/base/obj/mk/image.lds -MF build/base/obj/mk/image.lds.d -o build/base/obj/mk/image.lds mk/image.lds.S
EMBOX_GCC_LINK=full ./mk/extbld/arch-embox-gcc ./mk/extbld/toolchain_test.c -o ./build/base/obj/toolchain_test
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped```
```

As it was figured out, it's most likely the linker bug when performing relaxation (`ld --relax`): binutils-2.34/bfd/elf32-microblaze.c, at line 2223 (size of the next relax[i] is negative, and segfault occurs here).

_P.S. What is `--relax`?_
https://sourceware.org/binutils/docs/ld/Options.html
It performs a kind of target-specific optimizations.

anton-bondarev, 2020-04-05:
If I right remember`--relax` is microblaze specific flag for old gcc which fixed the problem with link with several linker scripts

alexkalmuk, 2020-04-05:
It's probably `LD_SINGLE_T_OPTION`.