embox/embox #1954: Characters get lost while copying a long (more then 7 symbols) string to the console

anton-bondarev, 2020-04-17:
To reproduce the problem you can
* run x86/qemu or arm/qemu template
* Copy to clipboard "1234567890"
* Paste to Embox command line
You see "12345678" instead of  "1234567890"

sobhaniiest, 2020-04-22:
Hi @anton-bondarev, I would like to work on this. Could you please assign this to me. Thanks :)

anton-bondarev, 2020-04-22:
@sobhaniiest I've assigned it for you

anton-bondarev, 2020-04-24:
@sobhaniiest It's enough to increase UART_DATA_BUFF_SZ in ttys_procesing.c for solving this issue. But it is not a good way I also want to rework "buffer implementation" part
```
struct uart_rx {
	struct uart *uart;
	int data;
	struct dlist_head lnk;
};

POOL_DEF(uart_rx_buff, struct uart_rx, UART_DATA_BUFF_SZ);
```
It is very exсess in my opinion

sobhaniiest, 2020-04-24:
> It's enough to increase UART_DATA_BUFF_SZ in ttys_procesing.c for solving this issue.

Thanks for helping me out. For now, should I create a PR by increasing `UART_DATA_BUFF_SZ` or work on the "buffer implementation" part?

anton-bondarev, 2020-04-27:
@sobhaniiest Can you investigate more deeply?
There is a loop in uart_irq_handler()
irq_return_t uart_irq_handler(unsigned int irq_nr, void *data) {
	struct uart *dev = data;

	if (dev->tty) {
		while (uart_hasrx(dev)) {
			uart_rx_buff_put(dev, uart_getc(dev));
		}
		lthread_launch(&uart_rx_irq_handler);
	}

	return IRQ_HANDLED;
}
 In which we check is there any symbols in UART and if there is get the symbols. Maybe before getting the character try to allocate object for processing, Now it's happened in uart_rx_buff_put()

sobhaniiest, 2020-04-27:
Sure! I'm working on this.

sobhaniiest, 2020-04-29:
I've tried to copy & paste different size strings. The loop is running exactly the total no of symbols in the string, but the allocation of objects in `uart_rx_buff_put()` is done only for `UART_DATA_BUFF_SZ` times. We need to know the size of the string (pasting) before the pool allocation.

anton-bondarev, 2020-04-29:
> I've tried to copy & paste different size strings. The loop is running exactly the total no of symbols in the string, but the allocation of objects in `uart_rx_buff_put()` is done only for `UART_DATA_BUFF_SZ` times. We need to know the size of the string (pasting) before the pool allocation.

@sobhaniiest I suggested in the[ comment](https://github.com/embox/embox/issues/1954#issuecomment-619788654) rework uart_rx_buff_put() and the loop at all.

I mean something like the following
```
if (dev->tty) {
	while (uart_hasrx(dev)) {
	irq_lock();
	{
		rx = pool_alloc(&uart_rx_buff);
		if (!rx) {
                   /* disable uart_irq */
                    uart_irq_disable(uart);
			irq_unlock();;
		lthread_launch(&uart_rx_irq_handler);
			return -1;
		}

		rx->uart = dev;
		rx->data = uart_getc(dev);
		dlist_add_prev(dlist_head_init(&rx->lnk), &uart_rx_list);
	}
	irq_unlock();
		lthread_launch(&uart_rx_irq_handler);
	}	
}
```
Of course, You have to disable irq when there are no free buffers and enable it if uart_rx_irq_handler  fulfilled (some free buffers will have appeared)

PR #1972 (2020-05-04): added uart_irq_dis/uart_irq_en to clear buffer
Closes #1954 
Thanks @anton-bondarev
closed 2020-05-04 by anton-bondarev