embox/embox #1955: readline (linenoise) has global data

anton-bondarev, 2020-04-17:
Linenoise has several globals
```
static linenoiseCompletionCallback *completionCallback = NULL;
static linenoiseHintsCallback *hintsCallback = NULL;
static linenoiseFreeHintsCallback *freeHintsCallback = NULL;

static struct termios orig_termios; /* In order to restore at exit.*/
static int maskmode = 0; /* Show "***" instead of input. For passwords. */
static int rawmode = 0; /* For atexit() function to check if restore is needed*/
static int mlmode = 0;  /* Multi line mode. Default is single line. */
static int atexit_registered = 0; /* Register atexit just 1 time. */
static int history_max_len = LINENOISE_DEFAULT_HISTORY_MAX_LEN;
static int history_len = 0;
static char **history = NULL;
```
so tish which used it must use sysmalloc and sysfree and history may contain trash line

puranjaymohan, 2020-04-23:
I want to work on this. Why are globals a problem? and what is sysmalloc?
Thanks

anton-bondarev, 2020-04-24:
The problem of global variables using is if Embox is configured without isolate process address spaces there would be only one instance of the variable for all tasks (processes) therefore we can not run several instance of this library at the same time. It's used in 'tish' (or other shells)  which in turn is used in telnet and so on. You can make sure about this, if connect with lelnet and look through command history There is single history fin Embox console and telnet.

sysmalloc is malloc which allocated memory from kernel task heap

@puranjaymohan It's better if you prepare PR for solving an issue (not only this issue). You can work on any issue you want or suggest your own ideas. It's opensource. :)
Issues (and PRs) are very useful for fixing (describing) the problem and discussions ways to solving it.
About this issue. It's rather complicated. Now I don't know how it can be solved.

puranjaymohan, 2020-04-24:
@anton-bondarev thanks for the explanation.
I am a beginner and that is why I ask before starting to solve an issue, I try to find issues which I am able to understand and solve.
Thanks.

anton-bondarev, 2020-04-27:
@puranjaymohan No problem :) Thanks for your contributions!

Look at #1914 and #1916 I think these more clean issues. You can try to solve it and of course, don't hesitate to ask an explanation and discuss the issue