embox/embox #1971: arm/qemu uses 100% one core of host cpu 100%

anton-bondarev, 2020-05-04:
It's enough to use arch_idle() with speeling

sk2sat, 2020-05-16:
What is "speeling"?
I think this issue can be fix by implementing arch_idle() for arm/qemu(arm926ejs).
Is this recognition correct?

And, does the detailed documentation for Mybuild exist?

anton-bondarev, 2020-05-16:
> I think this issue can be fix by implementing arch_idle() for arm/qemu(arm926ejs).
> Is this recognition correct?

I mean that 'arm/qemu' uses 'embox.arch.generic.arch' module
```
@Runlevel(0) include embox.arch.arm.arm9.arm926ejs.cpu -> depends embox.arch.generic.arch
```
'embox.arch.generic.arch' has trvial arch_idle implementation
```
void arch_idle(void) {
}
```

Therefore 'idle thread' is never asleep.
As I know there is an assembler instruction which makes CPU sleep until IRQ. you should use one in your arch_idle() implementation

anton-bondarev, 2020-05-16:
@sk2sat 

> What is "speeling"?
> I think this issue can be fix by implementing arch_idle() for arm/qemu(arm926ejs).
> Is this recognition correct?

Yes, I think You right, I described it above (previous comment).

> And, does the detailed documentation for Mybuild exist?

Unfortunately not, Some information can be found in issue #1774 and you can ask in the issue what exactly you want to know

sk2sat, 2020-05-16:
Thanks for detailed reply. I'll check it out.

PR #1996 (2020-06-04): Add arch_idle() for arm/qemu
Fix #1971
closed 2020-06-04 by anton-bondarev