embox/embox #2003: Improve SPI interface

anton-bondarev, 2020-05-29:
There are some examples of using SPI interface. Settings can be set up in mods.config and header, but in real life, each device has its required settings of SPI, for example, SPI clock polarity (high or low) frequently, how chip select must use, etc. These options can be set only in runtime before master or slave start use the bus

puranjaymohan, 2020-06-14:
I would like to work on this, please assign it to me.

anton-bondarev, 2020-06-15:
@puranjaymohan I've assigned.

Do you have any board to check SPI work?

puranjaymohan, 2020-06-15:
@anton-bondarev I have a STM32 nucleo board, I think it can be used for testing.

anton-bondarev, 2020-06-15:
> @anton-bondarev I have a STM32 nucleo board, I think it can be used for testing.

Great!

puranjaymohan, 2020-06-16:
@anton-bondarev I am trying to find the exact headers where SPI settings are present also the examples you said, where is it located.
thanks

anton-bondarev, 2020-06-16:
@puranjaymohan I mean that SPI settings are contained in [src/drivers/spi/stm32/spi_conf_f4.h](https://github.com/embox/embox/blob/master/src/drivers/spi/stm32/spi_conf_f4.h) and it's not very comfortable for users. It's better if users can set up pins in a template for their boards (project).

Let's try to start your board with SPI at first& What is exactly your board?

puranjaymohan, 2020-06-16:
@anton-bondarev I have ST's NUCLEO-L476RG board it is based on the STM32L476RG which is an ARM Cortex-M4 microcontroller. it has 3 SPIs and 1 Quad SPI.
https://www.st.com/en/evaluation-tools/nucleo-l476rg.html

anton-bondarev, 2020-06-16:
> @anton-bondarev I have ST's NUCLEO-L476RG board it is based on the STM32L476RG which is an ARM Cortex-M4 microcontroller. it has 3 SPIs and 1 Quad SPI.
> https://www.st.com/en/evaluation-tools/nucleo-l476rg.html

@puranjaymohan  Ok,  there is no support your board in Embox. But there is support  NUCLEO-F429ZI and stm32l4-discovery, therefore, I hope that porting will not difficult. Do you want to port Embox on  NUCLEO-L476RG. I will help you and try to buy the board as soon as possible. Should I create an issue to discuss the porting process or you do it?

puranjaymohan, 2020-06-16:
@anton-bondarev I would love to help in porting Embox to NUCLEO-L476RG. Please create the issue for the porting process and add the details as I never ported an RTOS to a new architecture. I think this would be a great project for me.

puranjaymohan, 2020-06-16:
It would be fun porting embox to the new microcontroller.

anton-bondarev, 2020-06-16:
> @anton-bondarev I would love to help in porting Embox to NUCLEO-L476RG. Please create the issue for the porting process and add the details as I never ported an RTOS to a new architecture. I think this would be a great project for me.

@puranjaymohan I've created issue #2034

puranjaymohan, 2020-06-21:
@anton-bondarev Now as it has been ported to Nucleo-L476RG, tell me how to proceed?

puranjaymohan, 2020-06-21:
@anton-bondarev I see that SPI is supported for stm32f4, so should we first make it independent of that or add SPI support for stm32l4?

anton-bondarev, 2020-06-21:
@puranjaymohan Do you have any devices with SPI interface which can be connected to Nucleo-L476RG?

> so should we first make it independent of that or add SPI support for stm32l4?

I think you should start with stm32l4 support. I'm going to improve SPI for stm32 within two days. Now I have nucleo-stm32f429 with connected a SPI device, and I can check it

puranjaymohan, 2020-06-21:
@anton-bondarev 
> Do you have any devices with SPI interface which can be connected to Nucleo-L476RG?
I have an Arduino, which I can use to send and receive dummy data over SPI after connecting it to Nucleo.

anton-bondarev, 2020-06-21:
> I have an Arduino, which I can use to send and receive dummy data over SPI after connecting it to Nucleo.

Great! Let's start with an example from cubel4?

puranjaymohan, 2020-06-22:
@anton-bondarev I tried running an example code from stm32cubel4 and I thing the following details from the example are useful for implementing the driver in embox

This is SPI1 related configurations
```
  hspi1.Instance = SPI1;
  hspi1.Init.Mode = SPI_MODE_SLAVE;
  hspi1.Init.Direction = SPI_DIRECTION_2LINES;
  hspi1.Init.DataSize = SPI_DATASIZE_8BIT;
  hspi1.Init.CLKPolarity = SPI_POLARITY_LOW;
  hspi1.Init.CLKPhase = SPI_PHASE_1EDGE;
  hspi1.Init.NSS = SPI_NSS_SOFT;
  hspi1.Init.FirstBit = SPI_FIRSTBIT_MSB;
  hspi1.Init.TIMode = SPI_TIMODE_DISABLE;
  hspi1.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
  hspi1.Init.CRCPolynomial = 7;
  hspi1.Init.CRCLength = SPI_CRC_LENGTH_DATASIZE;
  hspi1.Init.NSSPMode = SPI_NSS_PULSE_DISABLE;
```
This is for the GPIOs used by SPI1
```
    GPIO_InitTypeDef GPIO_InitStruct = {0};
    GPIO_InitStruct.Pin = GPIO_PIN_5|GPIO_PIN_6|GPIO_PIN_7;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF5_SPI1;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
```

anton-bondarev, 2020-06-23:
> @anton-bondarev I tried running an example code from stm32cubel4 and I thing the following details from the example are useful for implementing the driver in embox
>
> This is SPI1 related [...]

Yes, It seems you are in the right way!