embox/embox #2004: Improve GPIO API

anton-bondarev, 2020-05-29:
There is GPIO API [https://github.com/embox/embox/wiki/GPIO](https://github.com/embox/embox/wiki/GPIO) gpio_set(PORT, PIN, value).  But it would be useful if there was 'struct gpio' with port and pin number abstraction so could use gpio_pin_set(pin, value)

puranjaymohan, 2020-07-15:
@anton-bondarev Please explain more how this should work.

anton-bondarev, 2020-07-15:
@puranjaymohan 
I mean It's useful if it's expanded API with structure gpio wich contains port and pin fields (and controller of course). then it  possible to use the only pointer to gpio structure without store PORT and PIN value

puranjaymohan, 2020-07-15:
So we will use something like the following ?
```
static struct GPIOA mygpio;
gpio_pin_set(&mygpio->pin, 1);
```

anton-bondarev, 2020-07-15:
> So we will use something like the following?
> 
> ```
> static struct GPIO mygpio;
> mygpio.port = 'A';
> gpio_pin_set(&mygpio->pin1, 1);
> ```

Almost, but rather like that:
```
static struct gpio_pin gpio_pin;
 
gpio_pin.port = GPIO_PORT_A;
gpio_pin.pin = 1 << PIN_NUMBER;
gpio_pin_set(&gpio_pin, 1);
```

puranjaymohan, 2020-07-16:
@anton-bondarev I will start working on this, once I finish updating stm32cubef7.

anton-bondarev, 2020-07-16:
> @anton-bondarev I will start working on this, once I finish updating stm32cubef7.

Ok!

puranjaymohan, 2020-08-14:
Hi! @anton-bondarev I want to start working on it. I think it is a long process as all APIs have to change, how should I start working on it, without breaking anything.

anton-bondarev, 2020-08-15:
Hi, @puranjaymohan 

> I think it is a long process as all APIs have to change

Yes you are right. But you can not change, but extend the API, so you can  switch not all parts to the new API, it is enough to test the new API in one case (one board).

I suggest starting by checking the current state of the GPIO driver using the "pin" command. This will help you understand limitations current API. Let's discuss them.