embox/embox #2014: Run Embox on qemu risc-v

alexkalmuk, 2020-06-02:


sk2sat, 2020-06-12:
I am trying to create riscv/qemu64 template and fix src/arch/riscv/ to work on 64bit,
but is it better to create riscv64/qemu like mips64?

anton-bondarev, 2020-06-12:
> I am trying to create riscv/qemu64 template and fix src/arch/riscv/ to work on 64bit,
> but is it better to create riscv64/qemu like mips64?

I meant just run existing template riscv/qemu, but of cause, it's better if you create riscv64/qemu to start riscv64 supporting.
I think you should create full riscv64/qemu not only link as mips64

Is MAix BiT riscv64?

sk2sat, 2020-06-12:
Thanks.
Then I will implement riscv64/qemu(because MAix BiT is riscv64).

anton-bondarev, 2020-06-27:
Closed by #2053

anton-bondarev, 2020-06-27:
Closed by #2053

closed 2020-06-27 by anton-bondarev