embox/embox #2017: cortex-m: Store ISR vector in flash memory by default

alexkalmuk, 2020-06-02:
Currently, on almost all stm32 boards ISR vector is filled in RAM, which required an extra 512 bytes (if 128 entries), or even 1 Kb (256 entries). Practice shows us that RAM is a more "expensive" memory when we are trying to launch a lot of stuff in one config (many threads, sockets, large heap, etc.). So it looks like locating ISR vector in flash memory would be useful

puranjaymohan, 2020-07-14:
@alexkalmuk is this issue resolved? If not then I would like to work on this.

alexkalmuk, 2020-07-14:
Yes, it is resolved by https://github.com/embox/embox/pull/2046 and https://github.com/embox/embox/pull/2049. We have made a work on IRQ's there, so now ISR vector is located in ROM if you are using the static irqs (embox.kernel.irq_static).

closed 2020-07-14 by alexkalmuk