embox/embox #2034: Port on NUCLEO-L476RG

anton-bondarev, 2020-06-16:
Add supporting NUCLEO-L476RG board

anton-bondarev, 2020-06-16:
@puranjaymohan 
The closest board which Embox supports is arm/stm32l4-discovery. I think that it is a good idea to clone this template to start the issue.
Also, there is a sub-task enable stm32l4cube for different boards. Look at third-party/bsp/stm32f4cube or hird-party/bsp/stm32f4cube/stm32f7cube for example

puranjaymohan, 2020-06-16:
@anton-bondarev I saw the file third-party/bsp/stml4cube/Mybuild here there is not a separate directory for STM32L475, it should be in its separate directory and in the root directory only abstract modules should be defined as it is done for stm32f7cube.
So we will have to move STM32L475 specific modules to its own folder and create a new directory for STM32L576 and add its modules there.

Please tell me if my understanding is correct and then I can start working on this.

anton-bondarev, 2020-06-16:
> @anton-bondarev I saw the file third-party/bsp/stml4cube/Mybuild here there is not a separate directory for STM32L475, it should be in its separate directory and in the root directory only abstract [...]

@puranjaymohan Yes, you are right!

puranjaymohan, 2020-06-17:
@anton-bondarev I will try to first put STM32L475 in its separate folder and make the folder structure and then try to build the arm/stm32l4-discovery and see that I don't break it in the process.
Then I will copy the folder for STM32L475 BSP and create the BSP for STM32L476.

In the next step, I will copy arm/stm32l4-discovery and create the template for the NUCLEO board and then try to build it.
If all this goes fine then the stm32l4-discovery and Nucleo, both will work fine with their separate BSPs and Templates

anton-bondarev, 2020-06-17:
@puranjaymohan Yes, you in the right way

Also, I think you should check uart settings for STM32L476 in src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h. Of course, you can just run and look at the output in the terminal? but maybe your board has different settings (uart number and pin settings).

You can also check terminal settings if you run any working example (for example from cubeMX)

puranjaymohan, 2020-06-17:
@anton-bondarev I checked in the datasheet of my board, the TX pin is PA2 and RX pin is PA3 for the UART2, in src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h the values have been hardcoded for the discovery board, shouldn't the Serial Driver work regardless of the board type. or there should be a separate header for each board. The best would be if we can define the UART pins in the template itself.

anton-bondarev, 2020-06-17:
> @anton-bondarev I checked in the datasheet of my board, the TX pin is PA2 and RX pin is PA3 for the UART2, in src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h the values have been hardcoded for [...]

For now, I suggest using the same header but separate through UART number as in src/drivers/serial/stm32_usart/stm32_usart_conf_f4.h
there is branch `#elif MODOPS_USARTX == 2` for UART2 Nucleo and UART1 for discovery
Except this I suggest add a macro for board definition to CFLAGS (build.conf)

puranjaymohan, 2020-06-17:
@anton-bondarev I have added the following lines too, Will be making the PR in few minutes.
```
#define USART2_IRQ    \
	OPTION_MODULE_GET(embox__driver__serial__stm_usart_l4, NUMBER, usart2_irq)
static_assert(USART2_IRQ == USART2_IRQn);
```

puranjaymohan, 2020-06-18:
> Except this I suggest add a macro for board definition to CFLAGS (build.conf)

@anton-bondarev what did you mean by this? does it mean -DSTM32L476xx

anton-bondarev, 2020-06-18:
> > Except this I suggest add a macro for board definition to CFLAGS (build.conf)
> 
> @anton-bondarev what did you mean by this? does it mean -DSTM32L476xx

Yes I think it's a good idea to add -DSTM32L475xx and -DSTM32L476xx to CFLAGS instead of using @DefineMacro("STM32L475xx") and  @DefineMacro("STM32L476xx") in Mybuild file for stm32cubel4

PR #2043 (2020-06-21): Third-party: BSP: Port Embox to Nucleo-l476rg
Add the BSP and template for the ST's Nucleo-l476rg Board.
This closes #2034
It builds correctly but I have to test it on the board.
commit 17ed5829 (2020-06-21): Third-party: BSP: Port Embox to Nucleo-l476rg

Add the BSP and template for the ST's Nucleo-l476rg Board.
This closes #2034
closed 2020-06-21 by anton-bondarev