embox/embox #2035: Update stm32l4cube

anton-bondarev, 2020-06-16:
There is repository on github for stm32l4cube [https://github.com/STMicroelectronics/STM32CubeL4/releases/tag/v1.14.0](https://github.com/STMicroelectronics/STM32CubeL4/releases/tag/v1.14.0)

puranjaymohan, 2020-06-16:
@anton-bondarev 
I checked this file /home/puranjay/gsoc2020/embox/third-party/bsp/stml4cube/Makefile
PKG_SOURCES := http://download.embox.rocks/en.$(PKG_NAME).zip
so it downloads the file from here, does the zip file at this domain has to be changed to the latest zip file or the URL in this Makefile can be changed to https://github.com/STMicroelectronics/STM32CubeL4/archive/v1.14.0.zip
If this is the case then we can change it and make a pull request.

also, these links are not working http://download.embox.rocks/

anton-bondarev, 2020-06-16:
@puranjaymohan Thanks, You are right. I'm trying to fix the problem. I want to switch to downloading from https://github.com/STMicroelectronics/STM32CubeL4/archive/v1.14.0.zip

anton-bondarev, 2020-06-16:
> @puranjaymohan Thanks, You are right. I'm trying to fix the problem. I want to switch to downloading from https://github.com/STMicroelectronics/STM32CubeL4/archive/v1.14.0.zip

@puranjaymohan But if you do this it is great :)

puranjaymohan, 2020-06-16:
@anton-bondarev I have started working on it, I will make pull requests for all these issues soon.

PR #2038 (2020-06-17): Third-party: bsp: stml4cube: Update the download URL
Change the download URL to
https://github.com/STMicroelectronics/STM32CubeL4/archive/v1.14.0.zip so
that the latest file can be downloaded.
Change paths in the Mybuild to solve build errors that occur due to the new version.
closes #2035
closes #2039
closed 2020-06-17 by anton-bondarev