embox/embox #2037: Update stm32cubef7

anton-bondarev, 2020-06-16:
There is stm32cubef7 repository on github [ https://github.com/STMicroelectronics/STM32Cubef7]( https://github.com/STMicroelectronics/STM32Cubef7)

puranjaymohan, 2020-07-16:
I have started working on this.

anton-bondarev, 2020-07-16:
@puranjaymohan :+1:

puranjaymohan, 2020-07-17:
@anton-bondarev How to generate new patch files for this, namely, audio and patch.text

anton-bondarev, 2020-07-17:
> @anton-bondarev How to generate new patch files for this, namely, audio and patch.text

@puranjaymohan There is a helper script ./scripts/app_diff.sh. It contains comments how to use it.

puranjaymohan, 2020-07-19:
> > @anton-bondarev How to generate new patch files for this, namely, audio and patch.text
> 
> @puranjaymohan There is a helper script ./scripts/app_diff.sh. It contains comments how to use it.

@anton-bondarev I saw the script, but I am still confused. The old patch file had `STM32Cube_FW_F7_V1.5.0` at all places but the new archive doesn't have that folder, it has `STM32CubeF7-1.16.0`

alexkalmuk, 2020-07-20:
@puranjaymohan What if to replace `STM32Cube_FW_F7_V1.5.0` with new `STM32CubeF7-1.16.0` and check if these new patches will apply correctly?

puranjaymohan, 2020-07-21:
@alexkalmuk I tried that already, it is causing some problems.

anton-bondarev, 2020-07-24:
@puranjaymohan Could you create PR without patches? I'll try to create a new audio patch

anton-bondarev, 2020-07-25:
@puranjaymohan I think stmf4cube has a similar problem, there is no file for audio_patch
I described it in #2109

alexkalmuk, 2020-09-30:
Solved in https://github.com/embox/embox/pull/2205, thank you @anton-bondarev and @puranjaymohan !

What concerns audio.patch I removed it and checked that audio examples work without it well.

closed 2020-09-30 by alexkalmuk