embox/embox #2039: arm/stm32l4-discovery is broken

anton-bondarev, 2020-06-16:


puranjaymohan, 2020-06-16:
@anton-bondarev is this problem is due to the template or the BSP? I tried changing some paths in the BSP's Mybuild and arm/stm32l4-discovery built correctly.
I changed all references of `STM32Cube_FW_L4_V1.14.0` to `STM32CubeL4-1.14.0` in every place in the `embox/third-party/bsp/stml4cube/Mybuild `and after this it built properly.

anton-bondarev, 2020-06-17:
> I changed all references of `STM32Cube_FW_L4_V1.14.0` to `STM32CubeL4-1.14.0` in every place in the `embox/third-party/bsp/stml4cube/Mybuild `and after this it built properly.

@puranjaymohan Great. Add a commit with the changes to PR #2038
And you close both issues  #2039 and #2035

puranjaymohan, 2020-06-17:
@anton-bondarev I have made the changes and added the commit.
Please merge the PR, when you feel it is correct.

closed 2020-06-17 by anton-bondarev

puranjaymohan, 2020-06-17:
@anton-bondarev I was trying to build this and an error started to appear 
```error: 'OPTION_NUMBER_embox__driver__serial__stm_usart_f3__usart1_irq' undeclared here (not in a function); did you mean 'OPTION_NUMBER_embox__driver__serial__stm_usart_l4__usart1_irq'?```

What is this? how to solve this?

anton-bondarev, 2020-06-17:
@puranjaymohan Give me please more detailed log
`OPTION_NUMBER_embox__driver__serial__stm_usart_f3__usart1_irq`
It seems you've included 'stm_usart_f3' module instead of 'stm_usart_l4'

anton-bondarev, 2020-06-17:
@puranjaymohan Ops, I see.

puranjaymohan, 2020-06-17:
This line is in the header stm32_usart_conf_l4.h
```#define USART1_IRQ    \
     OPTION_MODULE_GET(embox__driver__serial__stm_usart_f3, NUMBER, usart1_irq)

puranjaymohan, 2020-06-17:
Why is this here, also earlier it was building fine, I don't know what happened now. Also, I didn't add that line there.

anton-bondarev, 2020-06-17:
```
diff --git a/src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h b/src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h
index 24dc6c3db..b5ea09df2 100644
--- a/src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h
+++ b/src/drivers/serial/stm32_usart/stm32_usart_conf_l4.h
@@ -25,7 +25,7 @@
 #define MODOPS_USARTX OPTION_GET(NUMBER, usartx)
 
 #define USART1_IRQ    \
-       OPTION_MODULE_GET(embox__driver__serial__stm_usart_f3, NUMBER, usart1_irq)
+       OPTION_MODULE_GET(embox__driver__serial__stm_usart_l4, NUMBER, usart1_irq)
 static_assert(USART1_IRQ == USART1_IRQn);
 
 #if MODOPS_USARTX == 1
```
Yes It's not your fault It happend in #2033

puranjaymohan, 2020-06-17:
@anton-bondarev So should, I add another PR for this or it will be solved by editing #2033

anton-bondarev, 2020-06-17:
> @anton-bondarev So should, I add another PR for this or it will be solved by editing #2033

Thanks a lot!

puranjaymohan, 2020-06-17:
@anton-bondarev I have added a PR for solving this, Now I can work on porting it to Nucleo Board :smile_cat:

PR #2038 (2020-06-17): Third-party: bsp: stml4cube: Update the download URL
Change the download URL to
https://github.com/STMicroelectronics/STM32CubeL4/archive/v1.14.0.zip so
that the latest file can be downloaded.
Change paths in the Mybuild to solve build errors that occur due to the new version.
closes #2035
closes #2039