embox/embox #2048: Build error

yuppox, 2020-06-20:
Using Debian Linux.
I am getting a build error:

```
for i in header_types.patch; do \
        patch -d /tmp/embox/build/extbld/third_party/lib/cjson -p0 < $PWD/$i; \
done
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff -aur cJSON.orig/cJSON.h cJSON/cJSON.h
|--- cJSON.orig/cJSON.h 2013-08-14 16:54:35.000000000 +0400
|+++ cJSON/cJSON.h      2014-07-11 13:39:02.904190163 +0400
--------------------------
```

alexkalmuk, 2020-06-21:
@yuppox Hi, which template you are building? Are you in /tmp/embox folder?

anton-bondarev, 2020-06-21:
@yuppox Please give previous output lines.

I met this message when I forgot to install unzip

alexkalmuk, 2020-09-29:
Please reopen if you still observe the problem.

closed 2020-09-29 by alexkalmuk