embox/embox #2054: GDB reports corrupt stack when debugging x86/qemu

abusalimov, 2020-06-23:
```
Breakpoint 1, __schedule (preempt=22) at src/kernel/sched/sched.c:374
374		prev = schedee_get_current();
(gdb) bt
#0  __schedule (preempt=22) at src/kernel/sched/sched.c:374
#1  0x00000016 in ?? ()
#2  0x00000016 in ?? ()
#3  0x0015939d in __critical_count_add (count=1) at ./src/include/kernel/critical.h:150
#4  0x0015a271 in sched_preempt () at src/kernel/sched/sched.c:428
#5  0x00000001 in ?? ()
#6  0x0003f000 in ?? ()
#7  0x01b9ae28 in _bss_end ()
#8  0x0015a267 in sched_preempt () at src/kernel/sched/sched.c:427
#9  0x00000016 in ?? ()
#10 0x00000216 in ?? ()
#11 0x01b9ae58 in _bss_end ()
#12 0x0010af18 in critical_dispatch_pending () at src/kernel/critical.c:35
#13 0x01b9ae54 in _bss_end ()
#14 0xfffc0000 in ?? ()
#15 0x00000016 in ?? ()
#16 0x0003ffff in ?? ()
#17 0x0024cd8c in clock_source_list ()
#18 0x00000216 in ?? ()
#19 0x00000000 in ?? ()
```

### Steps to reproduce:
- build x86/qemu
- run `auto_gemu -s -S`
- attach with GDB to `:1234`
- put a breakpoint to some function, I tried in `__schedule`, and into a test case of 'critical'.
- when the breakpoint hits, observe the corrupt stack
- try next on a line with a call, it steps into instead of stepping over the call

Adding `-fno-omit-frame-pointer` has no effect.

WSL2, GCC 9.3.0, GDB 9.1, QEMU 4.2.0

alexkalmuk, 2020-09-16:
Hi! Today I faced the same problem after installing the recent ubuntu 20.04.

I checked with gcc7 that gdb works fine, but starting with gcc8 (8.4.0 for me) gdb reports corrupt stack. Examing disassembler I found `endbr32` (it's part of recent Control-flow Enforcement Technology (CET)) inserted at almost every function prologue:

```asm
00151b5d <__schedule>:
  151b5d:   f3 0f 1e fb             endbr32
  151b61:   55                      push   %ebp
  151b62:   89 e5                   mov    %esp,%ebp
  151b64:   83 ec 18                sub    $0x18,%esp
```

I wondering how this leads to a corrupt stack because I got the same `endbr`'s compiling "hello world" on my host system.

Launching qemu with `-d in_asm` you can see how instructions were interpreted:
```
./scripts/qemu/auto_qemu -no-kvm -d in_asm -s -S
```

For instance, 

```
gdb) b arch_init
Breakpoint 1 at 0x101bc7: file /src/arch/x86/kernel/arch.c, line 34.
(gdb) target remote :1234
Remote debugging using :1234
0x0000fff0 in ?? ()
(gdb) c
Continuing.

Breakpoint 1, arch_init () at /src/arch/x86/kernel/arch.c:34
34	void arch_init(void) {
(gdb) disassemble 
Dump of assembler code for function arch_init:
=> 0x00101bc7 <+0>:	endbr32 
   0x00101bcb <+4>:	push   %ebp
   0x00101bcc <+5>:	mov    %esp,%ebp
   0x00101bce <+7>:	nop
   0x00101bcf <+8>:	pop    %ebp
   0x00101bd0 <+9>:	ret    
End of assembler dump.
(gdb) si
```

will show you in QEMU:

```
----------------
IN: 
0x00139981:  f3                       .byte    0xf3
0x00139982:  0f                       .byte    0x0f
0x00139983:  1e                       pushl    %ds
0x00139984:  fb                       sti      
```

So you can see qemu do not interpret `endbr32` correctly. I cannot explain in more detail, but probably it can be a root of the problem.

Disabling `endbr32` generation solves the problem for me. It's made in https://github.com/embox/embox/pull/2182

alexkalmuk, 2020-09-16:
btw, I' now looking towards `-fcf-protection=none` instead of `-mmanual-endbr`.

alexkalmuk, 2020-09-17:
Fixed in https://github.com/embox/embox/pull/2182. Please, reopen this issue if you will observe the problem.

closed 2020-09-17 by alexkalmuk