embox/embox #2055: Add SPI driver support for STM32Cubel4

puranjaymohan, 2020-06-23:
Add support in the SPI driver for microcontrollers based on STM32CubeL4. Currently, Embox supports two boards based on L4, STM32l4-discovery, and NUCLEO-l476RG.
Currently, the SPI driver only supports stm32cubef4 and therefore we need to add support for other boards too.

puranjaymohan, 2020-06-23:
I have mostly completed adding the L4 support in SPI driver, but the SPI driver uses the GPIO driver which is not present for L4, so the GPIO driver has to be created for L4 then only SPI can be added.

puranjaymohan, 2020-06-24:
@anton-bondarev I am getting an error while compiling the SPI driver, can you tell me what it is.
```
In file included from build/base/include/drivers/spi.h:14,
                 from src/drivers/spi/stm32/stm32_spi1.c:10:
build/base/include/drivers/spi.h:61:37: error: initialization discards 'const' qualifier from pointer target type [-Werror=discarded-qualifiers]
   61 |  CHAR_DEV_DEF(dev_name, NULL, NULL, &spi_iops, &MACRO_CONCAT(spi_device, idx)) \
      |                                     ^
build/base/include/drivers/char_dev.h:23:38: note: in definition of macro 'CHAR_DEV_DEF'
   23 |    .o = open_fn, .c = close_fn, .i = idesc_op, .p = priv \
      |                                      ^~~~~~~~
src/drivers/spi/stm32/stm32_spi1.c:78:1: note: in expansion of macro 'SPI_DEV_DEF'
   78 | SPI_DEV_DEF("stm32_spi_1", &stm32_spi_ops, &stm32_spi1, 1);
```

anton-bondarev, 2020-06-24:
@puranjaymohan  It seems the problem is in CHAR_DEV_DEF macro implementation without devfs . Try to add casting to (void *)  
https://github.com/embox/embox/blob/master/src/drivers/char/char_dev_stub.h#L23 
```
.o = open_fn, .c = close_fn, .i = (void *)idesc_op, .p = priv \
```
And don't forget 'make clean' before rebuild

puranjaymohan, 2020-06-24:
@anton-bondarev Now it builds correctly after adding (void *). Please merge the GPIO PR because SPI depends upon GPIO and then only I will be able to add PR for SPI.

anton-bondarev, 2020-06-24:
> Please merge the GPIO PR

@puranjaymohan done!

puranjaymohan, 2020-07-14:
@anton-bondarev #2058 closes this.

closed 2020-07-15 by puranjaymohan

anton-bondarev, 2020-07-15:
> > Please merge the GPIO PR
> 
> @puranjaymohan done!

O, yes, of course, thanks!