embox/embox #2066: Detect STM32F4-Discovery as USB device using lsusb

ThePiGrepper, 2020-06-29:


alexkalmuk, 2020-07-01:
How are you going to start? I'd propose to add a driver in Embox based on STM32Cube directly. Then we can reply something there for "get device descriptor" request.

ThePiGrepper, 2020-07-02:
> How are you going to start? I'd propose to add a driver in Embox based on STM32Cube directly. Then we can reply something there for "get device descriptor" request.

I was doing this directly. however I took a look at the new gadget structure you merged today and I think it's better to start testing that interface with as many gadgets as possible, as well as with more than one UDC, to see if it needs some changes. I'm currently working on the UDC implementation for stm32f4 based on CubeF4, and on a CDC gadget. I was going to do the CDC gadget first, but I dont have a da14695 so I wouldnt be able to test it ASAP. I'll be posting this code during the day.

alexkalmuk, 2020-07-03:
>  I'm currently working on the UDC implementation for stm32f4 based on CubeF4

:+1: 

> I was going to do the CDC gadget first, but I dont have a da14695 so I wouldnt be able to test it ASAP. I'll be posting this code during the day.

Yes, if you will have a CDC gadget, you can push it to a branch and we will try to check. There is only RNDIS gadget now in repo, and it's very similar to CDC ECM. By the way, you can create branches directly in this repository.

ThePiGrepper, 2020-07-03:
> p.s. By the way, you can create branches directly in this repository, if this would be handy for you. It is also handy for us to verify changes in local branches.
Oh, I see. that could be helpful indeed. thx for letting me know.

alexkalmuk, 2020-09-15:
Closed in https://github.com/embox/embox/pull/2166

closed 2020-09-15 by alexkalmuk