embox/embox #2076: USB device mode discussion

alexkalmuk, 2020-07-01:
USB device mode (in Linux it's called USB gadget) allows you to create USB devices like CDC, MSC, HID, etc. There is no such ability in Embox yet (but we are working on it now). But what is relatively (but not fully) clear to me currently:

* There should be USB device controller (UDC). It's a hardware controller that performs all low-level work - enables endpoints, moves USB data to and from endpoints, etc. It was figured out that UDC also handles resets, suspends. It also handles SET_ADDRESS. Most other USB requests pass further to higher levels.
* There should be device USB interface (or USB function). A single interface performs a single function - for example, MSC or CDC. Function handles requests passed from UDC  which were sent to the interface from a host.
* Finally, gadget is a combination of interfaces. In my mind, it is "device descriptor" + "configuration descriptor" + USB functions.

alexkalmuk, 2020-07-01:
I added one UDC for da14695 (cortex-m33), but we need more UDC examples to better understand how controllers work. I think it's a good point to investigate is https://github.com/embox/embox/issues/2066

ThePiGrepper, 2020-07-02:
> I added a UDC driver for da14695 (cortex-m33), but we need more UDC examples to understand better how controllers work in general. I think it's a good point to investigate is #2066

indeed. I took my time reading the new code from https://github.com/embox/embox/pull/2075, and I basically understood it, so I'm trying to get the UDC for STM32F4 done today.

alexkalmuk, 2020-07-24:
I added a simple CDC-ECM module (very basic draft version) in https://github.com/embox/embox/pull/2106. It's a draft version, but with a working network (ping, dhcpd, telnetd) on da14695. But currently, it can be also used for debugging purposes to see lsusb.