embox/embox #2079: sifive-plic haven't been implemented yet

anton-bondarev, 2020-07-01:


SurajSonawane2415, 2024-05-28:
Hello @anton-bondarev,

I want to start working on this issue. I have studied the essentials for PLIC implementation for sifive, including the memory map, interrupt handling process, and registers in detail from the [SiFive U54-MC Core Complex Manual v1p0](https://static.dev.sifive.com/U54-MC-RVCoreIP.pdf). I have also added more information about this in the proposal I sent you via email.

I am planning to add the following components:
```
src/
  drivers/
    interrupt/
      sifive_plic/
        ├── mybuild
        ├── sifive_plic.h
        └── sifive_plic.c
```

Could you please guide me on how to start?

anton-bondarev, 2024-05-28:
@SurajSonawane2415 Thanks.

> I am planning to add the following components:

Yes it looks OK

> Could you please guide me on how to start?

- I suggest to start with runnign a sifive platform on qemu.
- Then look at src/drivers/interrupt/riscv_plic/riscv_plic.c to study it.
- Add a stub for your new driver
- And finally implemente it

PS I've seen your email. I will answer asap.

SurajSonawane2415, 2024-05-28:
Thank you for your response. I will start working on it.

> @SurajSonawane2415 Thanks.
>
>
> Yes it looks OK
>
>
> * I suggest to start with runnign a sifive platform on qemu.
> * Then look at src/drivers/interrupt/riscv_plic/riscv_plic.c to study it.
> * Add a stub for [...]

Thank you for your response. I will start working on it.

SurajSonawane2415, 2024-06-08:
@anton-bondarev I have implemented code for sifive-plic. You can find the code at https://github.com/SurajSonawane2415/embox/tree/master/src/drivers/interrupt/sifive_plic. Can you please review and suggest any necessary changes?  
I referred to the SiFive U54-MC Core Complex Manual v1p0 at https://static.dev.sifive.com/U54-MC-RVCoreIP.pdf.

Additionally, I have a question regarding the handling of hart IDs. The SiFive U54-MC Core Complex has 5 harts. In my current implementation, I have assumed `hart_id = 0`. However, I am planning to modify the function to take the hart ID as a parameter, like this:

```c
void irqctrl_enable(uint32_t hart_id, unsigned int interrupt_nr);
```

However, this function is already defined and used in many interrupt codes. Can you suggest how I can approach taking the hart_id?

Thank you!

anton-bondarev, 2024-06-08:
@SurajSonawane2415 Thanks a lot!

> I have implemented code for sifive-plic. You can find the code at https://github.com/SurajSonawane2415/embox/tree/master/src/drivers/interrupt/sifive_plic. Can you please review and suggest any necessary changes?

It looks good at the first glance. Please create a new PR with with changes

> ```c
> void irqctrl_enable(uint32_t hart_id, unsigned int interrupt_nr);
> ```

I think you should either use cpuid() for getting hart_id or extend th irqctrl API (with something like
 `void irqctrl_enable_in_cpu(uint32_t hart_id, unsigned int interrupt_nr);`

SurajSonawane2415, 2024-06-08:
@anton-bondarev thank you I will make changes and create a PR.

anton-bondarev, 2024-06-10:
Closed #3314

closed 2024-06-10 by anton-bondarev